OctoPrint/octoprint-docker #2: Pip command does not work correctly for this installation of OctoPrint.

armicron, 2017-01-30:
I will try to fix it maybe.
[image]

gaetancollaud, 2017-01-30:
Never had the chance to install plugins inside this docker container. It's strange because pip is installed. It comes with Python:2.7 : https://github.com/docker-library/python/blob/a248f4583c5f788e1af02016f762cdc323ee5765/2.7/Dockerfile

Maybe it has something to do with permissions since this container run using the _octoprint_ user.

hypergig, 2017-01-31:
@gaetancollaud it isn't that
```
talisker:~ jordan$ 
talisker:~ jordan$ docker exec -it octoprint bash
octoprint@1d12e2176e7f:/opt/octoprint$                                                                                                                      
octoprint@1d12e2176e7f:/opt/octoprint$ pip --version
pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)
octoprint@1d12e2176e7f:/opt/octoprint$ 
octoprint@1d12e2176e7f:/opt/octoprint$ whoami
octoprint
octoprint@1d12e2176e7f:/opt/octoprint$ 
octoprint@1d12e2176e7f:/opt/octoprint$ exit
talisker:~ jordan$ 
talisker:~ jordan$ 
talisker:~ jordan$ docker exec --user root -it octoprint bash
root@1d12e2176e7f:/opt/octoprint#                                                                                                                           
root@1d12e2176e7f:/opt/octoprint# pip --version
pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)
root@1d12e2176e7f:/opt/octoprint# 
root@1d12e2176e7f:/opt/octoprint# whoami
root
root@1d12e2176e7f:/opt/octoprint# 
root@1d12e2176e7f:/opt/octoprint# exit
talisker:~ jordan$ 
```
I suspect a `path` issue here.. 

I am diggin through the code now to better understand how octoprint calls `pip`.

gaetancollaud, 2017-02-11:
Thanks @hypergig Just to be sure I ran octoprint as root but same result as you would expect.

I think the problem comme from the [pip.py](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/pip.py#L104) file. The command is None, like there is no pip available. I will open a issue in the main project, maybe someone has an idea.

gaetancollaud, 2017-02-13:
As requested by @foosel by mail here is the log when octoprint starts.

```log
octoprint    | 2017-02-13 19:04:03,081 - octoprint.server - INFO - Starting OctoPrint 1.3.0.post1+g1014712 (master branch)
octoprint    | 2017-02-13 19:04:03,105 - octoprint.plugin.core - INFO - Loading plugins from /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins, /home/octoprint/.octoprint/plugins and installed plugin packages...
octoprint    | 2017-02-13 19:04:03,613 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
octoprint    | 2017-02-13 19:04:03,614 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
octoprint    | 2017-02-13 19:04:03,626 - octoprint.plugin.core - INFO - Loading plugins from /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins, /home/octoprint/.octoprint/plugins and installed plugin packages...
octoprint    | 2017-02-13 19:04:03,725 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
octoprint    | 2017-02-13 19:04:03,727 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/octoprint/.octoprint/uploads...
octoprint    | 2017-02-13 19:04:03,728 - octoprint.filemanager.storage - INFO - ... file metadata for /home/octoprint/.octoprint/uploads initialized successfully.
octoprint    | 2017-02-13 19:04:03,736 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
octoprint    | 192.168.192.113 - - [13/Feb/2017 19:04:04] "GET /sockjs/info?t=1487012643854 HTTP/1.1" 404 -
octoprint    | 2017-02-13 19:04:04,224 - octoprint.util.pip - INFO - Using "/opt/octoprint/venv/bin/python -m pip" as command to invoke pip
octoprint    | 2017-02-13 19:04:04,224 - octoprint.util.pip - DEBUG - Going to figure out pip's version
octoprint    | 2017-02-13 19:04:04,662 - octoprint.util.pip - INFO - Version of pip is 9.0.1
octoprint    | 2017-02-13 19:04:04,662 - octoprint.util.pip - ERROR - Cannot use pip
octoprint    | 2017-02-13 19:04:04,663 - octoprint.plugin.core - INFO - Initialized 7 plugin implementation(s)
octoprint    | 2017-02-13 19:04:04,665 - octoprint.plugin.core - INFO - 7 plugin(s) registered with the system:
octoprint    | |  Announcement Plugin (bundled) = /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins/announcements
octoprint    | |  Core Wizard (bundled) = /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins/corewizard
octoprint    | |  CuraEngine (<= 15.04) (bundled) = /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins/cura
octoprint    | |  Discovery (bundled) = /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins/discovery
octoprint    | |  Plugin Manager (bundled) = /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins/pluginmanager
octoprint    | |  Software Update (bundled) = /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins/softwareupdate
octoprint    | |  Virtual Printer (bundled) = /opt/octoprint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0.post1+g1014712-py2.7.egg/octoprint/plugins/virtual_printer
octoprint    | 2017-02-13 19:04:04,666 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
octoprint    | 2017-02-13 19:04:04,669 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
octoprint    | 2017-02-13 19:04:04,671 - octoprint.server - INFO - Reset webasset folder /home/octoprint/.octoprint/generated/webassets...
octoprint    | 2017-02-13 19:04:04,676 - octoprint.server - INFO - Reset webasset folder /home/octoprint/.octoprint/generated/.webassets-cache...
octoprint    | 2017-02-13 19:04:05,087 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on 22ab1dc201ba for SSDP
octoprint    | 2017-02-13 19:04:05,091 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
octoprint    | 2017-02-13 19:04:05,153 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
octoprint    | 2017-02-13 19:04:05,169 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.192.113
octoprint    | 2017-02-13 19:04:05,237 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.192.69:5000/'}
```

We can clearly see ` ERROR - Cannot use pip` but I don't know were it come from. I enabled the log debug for `octoprint.util.pip`. Tell me if something else is useful ?
```yaml
loggers:
  octoprint.util.pip:
    level: DEBUG
```

If it can help here is the beginning of the `api/plugin/pluginmanager` call.
```JSON
{
  "octoprint": "1.3.0.post1+g1014712",
  "os": "linux",
  "pip": {
    "additional_args": null,
    "available": false,
    "install_dir": null,
    "python": "\/opt\/octoprint\/venv\/bin\/python",
    "use_user": false,
    "version": null,
    "virtual_env": false
  },
 //...
}
```

foosel, 2017-02-13:
My lack of docker experience might be showing here again (and the fact that I haven't yet gotten around to testing this myself) but: is the site-packages folder of the virtual environment writable by the OctoPrint process in this setup? Because this kind of output can only be produced if the check here fails: https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/pip.py#L360. And if OctoPrint can't write to the package folder associated with the python environment it also can't install packages and if it can't do that it also can't install plugins (because those are packages) 

If this is in fact the problem, it might be time for me to look closer into allowing a secondary python packages folder in which to install plugins, which could then reside on the writable part of the container.

hypergig, 2017-02-14:
@foosel where is the 'package folder associated with the python environment' ?

armicron, 2017-02-14:
May I answer? It's in `/opt/octoprint` directory. 
```
$ ls -l /opt/octoprint/
total 388
...
drwxr-xr-x  5 octoprint octoprint   4096 Feb 14 06:40 venv
```

gaetancollaud, 2017-02-14:
@armicron Thanks for PR, it fixed the issue for me, I can now install plugins.

In the meantime I switched from gaetancollaud/octoprint to octoprint/octoprint

closed 2017-02-14 by gaetancollaud

hypergig, 2017-02-15:
awesome work!

CoderFX, 2018-09-16:
fix this damn thing!

jel111, 2020-02-08:
How was this fixed I have the same issue. From reading above I don't understsand what fixed this.