OctoPrint/octoprint-docker #6: ARM builds

y8, 2018-03-12:
Hello!

It would be nice to have automatic builds for armhf arch as well. 

SEE: https://resin.io/blog/building-arm-containers-on-any-x86-machine-even-dockerhub/

Cheers!

cmgrass, 2018-12-28:
Not sure if this is related, but:

pi@mypi:~/octoprint-docker $ docker-compose up -d
Starting octoprint
pi@mypi:~/octoprint-docker $ docker-compose logs -f
Attaching to octoprint
octoprint    | standard_init_linux.go:190: exec user process caused "exec format error"
octoprint exited with code 1

I assume this is because the raspberry pi is ARM

I will try to build a container

williamwebb, 2019-07-13:
To this point, this should be turned into a [multi-arthicture image](https://blog.docker.com/2017/09/docker-official-images-now-multi-platform/)

yamlCase, 2019-09-17:
inspecting the Dockerfile, the parent image `FROM python:2.7` does support multi architecture including arm.  My guess is the main code README.md will work for docker-compose as it is building the image prior to run.  

For the "without docker-compose" documentation, you'll need to build it first: `docker build -t octoprint:pi` then in the run command replace "octoprint/octoprint" with "octoprint:pi"

gaetancollaud, 2019-09-18:
Pull requests are welcome. I'm using a x86 architecture to run it so I cannot really test it.

LongLiveCHIEF, 2020-03-30:
Now that we have multi-arch image builds in place, this is the next thing I'm implementing. Currently we're building for `amd64` and `arm64`, but I plan to immediately implement support for any architectures that the ffmpeg static binaries can be installed on, including `armhf` and `armel`

Look for this in the next day or two.

PR #30 (2020-03-30): add support and builds for armhf
Limited to the architecture platform the python base image supports, so only able to add armv7 at this time.

closes #6
closed 2020-03-30 by LongLiveCHIEF