OctoPrint/octoprint-docker #7: Permisions on /dev/ttyACM0

JoschaMiddendorf, 2018-03-24:
I mapped ttyACM0 to the container by "--device /dev/ttyACM0:/dev/ttyACM0".
ttyACM0 is present in the container as follows:
"crw-rw---- 1 root root 166, 0 Mar 24 00:42 ttyACM0"

Is seems as OctoPrint hasn't the right permissions to access ttyACM0 as I'm getting the following error on my attempt to connect.

Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: /dev/ttyACM0 OSError: '[Errno 13] Permission denied: '/dev/ttyACM0'' @ comm.py:_openSerial:1931 (hook default)

gaetancollaud, 2018-03-24:
Have you tried to start the container with the `--privileged` flag. Maybe your docker user doesn't have the rights for the serial devices. See documentation here: https://docs.docker.com/engine/reference/run/#security-configuration

Otherwise you can try to add the docker user to the `dialout` group.

JoschaMiddendorf, 2018-03-24:
`--privileged` doesn't helped either. As the container is runing on a UnRaid server date device belongs to root:root.
I tried starting the container with `--security-opt="label=user:root"`, but that didn't helped either. 
How can I allow OctoPrint inside the container to access `crw-rw---- 1 root root 166, 0 Mar 24 00:42 ttyACM0`

gaetancollaud, 2018-03-25:
I think your issue is more related to docker and how device are "mounted" and not the ocotprint image itselft.

In the documentation they say you can can have attributes for devices, can you try to change them ? https://docs.docker.com/engine/reference/commandline/run/#add-host-device-to-container---device

Also are you using linux as host ? I had issue with Windows and devices.

jmerdich, 2018-04-13:
So, the octoprint user inside the container has the 'dialout' group (GID 20). The user or group must have permission to read and write from the device on the host, and only your root user has that, which is where the permissions issue stems from. 

Mapped file permissions are matched between the user and group id numbers *of the user running inside the container* and *on the host file system* (Docker's `--device` is mostly just a file mapping). In this case, the filesystem only allows access to the root user (UID 1) and root group (GID 1), whereas the octoprint instance is [running as an in-container octoprint user (UID ???) and dialout group (GID 20)](https://github.com/OctoPrint/docker/blob/master/Dockerfile#L37). We can fix this by changing the permissions on our 3d printer's connection to allow the dialout group (you can also fix it by running octoprint as root or with the root group inside the container but that's a horrible idea).  

### Enough explanation, fix it
1. Ensure you have the dialout group on your host: `$ grep dialout /etc/group` should start with `dialout:x:20:`. If you don't have it: `# groupadd -g 20 dialout`
2. Make the device be owned by dialout: `# chown root:dialout /dev/ttyACM0`. This won't last after the printer is unplugged.
3. Test the container. 
4. If it works, make it persist across unplugs/reboots with the following udev rule. Modify the values for the vendor and product ids with the ones on your 3d printer, found with `lsusb` (eg: `Bus 001 Device 005 ID 27b1:0001 Mr 3D Printer`).
```
# file: /etc/udev/rules.d/99_octoprint
# ids here are for for RAMBO, get your own if that's not you!
SUBSYSTEM="tty", ATTRS{idVendor}=="27b1", ATTRS{idProduct}=="0001", MODE="0660", GROUP="dialout"
```
5. Reload the rules: `# udevadm control --reload-rules && udevadm trigger` (you can also just reboot)
6. Unplug, wait a bit, and replug the 3d printer in, then check the group of `/dev/ttyACM0`

gaetancollaud, 2018-04-16:
Thanks for your explanation @jmerdich! 

@JoschaMiddendorf can you confirm that it works ? If so I will add a troubleshooting page in the wiki.

michaelwoods, 2018-07-25:
Adding the device and my user to dialout did not solve this issue for me. I still cannot open the device from within the container, privileged or not.

```
Unexpected error while connecting to serial port: /dev/ttyUSB0 SerialException: '[Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'' @ comm.py:_openSerial:2308 (hook default)
```

This seems related, though I'm only running docker v17.12.1-ce: https://github.com/docker/for-linux/issues/228

gaetancollaud, 2018-07-25:
Thanks @michaelwoods I will close this

closed 2018-07-25 by gaetancollaud

ElwayDenmark, 2018-12-23:
@jmerdich   anyway to get in touch with you? I have a similar issue (I believe) with Home Assistant in a docker container not connect with a z-wave usb stick. I try to follow your instructions, but cannot quite get it to work (linux noob). I'll buy you a beer or two if you can help me out :-)