OctoPrint/octoprint-docker #8: failes ffmpeg url

xpix, 2018-05-17:
Please check:

```
--2018-05-17 09:43:27--  https://johnvansickle.com/ffmpeg/releases/ffmpeg-release-32bit-static.tar.xz
Resolving johnvansickle.com (johnvansickle.com)... 162.222.226.121
Connecting to johnvansickle.com (johnvansickle.com)|162.222.226.121|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-05-17 09:43:29 ERROR 404: Not Found.

```

Cthanatos, 2018-08-29:
Well, until it gets fixed - here's what I did. (use at your own risk, as SSL certs are important for a reason)

- Get into the shell of your docker as root user with
`docker exec -u 0 -it organizr /bin/bash`

 - change to tmp directory
 `cd /tmp/`

- Use wget to get the file, but you have to ignore the certificate issues ( You'll get warnings, but it still downloads)
` wget --no-check-certificate -O  ffmpeg.tar.xz https://johnvansickle.com/ffmpeg/releases/ffmpeg-release-32bit-static.tar.xz`

 - Make the directory before unzipping the tar
` mkdir -p /opt/ffmpeg`

 - Untar to the right location
`tar xvf ffmpeg.tar.xz -C /opt/ffmpeg --strip-components=1`

 - Remove ffmpeg download file
`rm -Rf /tmp/*`

 - Set right location in webUI when setting up for ffmpeg
`/opt/ffmpeg/ffmpeg`

fionnb, 2019-02-18:
The new, correct URL is:
https://johnvansickle.com/ffmpeg/releases/ffmpeg-release-armel-static.tar.xz

It is just a matter of replacing it in the Dockerfile.

PR #27 (2020-03-30): Multi arch build
Switches from travis-ci to github actions. Will require secrets to be added in settings > secrets for `DOCKER_USERNAME` and  `DOCKER_PASSWORD`.

fixes #21
fixes #8 #18
closed 2020-03-30 by LongLiveCHIEF