OctoPrint/octoprint-docker #10: Persistent Configuration and Plugins

JoschaMiddendorf, 2018-06-01:
Which volumes do I have to mapp outside the container to make my configuration and installed plugins persistent across updates?

gaetancollaud, 2018-06-09:
Normally `/home/octoprint/.octoprint` should be enough. Do you use the [docker-compose](https://github.com/OctoPrint/docker/blob/master/docker-compose.yml) file ?

JoschaMiddendorf, 2018-06-09:
No as I’m using UnRaid OS docker compose isn’t supported. I’m using a huge generated docker run command and with only .octoprint mounted all plugins are gone after any updates.

stickystyle, 2018-12-28:
I just came across this as well, the issue is that plugins that are installed via the plugin manager get put into the venv where Octoprint is running, namely `/opt/octoprint/venv/lib/python2.7/site-packages/` and this dir isn't exported as a volume (I'm not sure you would want to).
I've only been playing with Octoprint for the last 8 hours, but I'm thinking perhaps passing `-t /home/octoprint/.octoprint/plugins` option to PIP from the plugin manager should get the plugins into the volume so they are not lost when the container is recreated.

TheFitzZZ, 2019-01-30:
@stickystyle could you please elaborate on the parameter passing you mentioned?

DanielJoyce, 2019-02-03:
No need for that. Just need to configure the docker-compose volumes properly.

TheFitzZZ, 2019-02-06:
@DanielJoyce we already established that this is not the case.

balert, 2019-03-14:
I've had the same problem and found a really hacky solution for it. Seems to work for me. 
The trick was similar to what @stickystyle proposed.

Well, -t /home/octoprint/.octoprint/plugins didn't work for me. Once I've figured out you can pass pip options in the octoprint settings under plugin manager I've tried this, and it didn't work because the plugin folder was empty and the installation of a plugin required the "OctoPrint" package as a dependency. Fiddled around a bit with this but it didn't work.

Then I thought I might just mount the site-package directory, as also proposed by @stickystyle but this didn't work either because it needed to be populated first, and coulnd't be empty. 

So what I did so solve this was to mount a volume "venv" from outside to "/venv" inside the container and then get a bash inside the container (docker-compose exec octoprint bash) and copy the contents of /opt/octoprint/venv to /venv. Then I stopped the container and mounted the outside "venv" directory to /opt/octoprint/venv, by this octoprint resides outside the container and plugins are persistent and work. 

However, clearly this isn't a good solution, I'm happy about any hints or tips to get this working.

JoschaMiddendorf, 2019-03-14:
That’s exactly what I did as a workaround.

balert, 2019-03-14:
> That’s exactly what I did as a workaround.

nice,  I just thought I could comment this here, so others can save time, and might help.

JoschaMiddendorf, 2019-03-15:
Just to be exact. I just mounted /opt/octoprint/venv/lib/python2.7/site-packages/ not the whole /opt/octoprint/venv.

TheFitzZZ, 2019-03-15:
That's what I did a few months ago - unfortunately that broke later on, as updates within the updated docker image itself wrecked it. So you could argue: don't update the image, always update through the updater :-)

balert, 2019-03-15:
@JoschaMiddendorf in my case only site-packages didn't suffice. Don't recall the details.

johny-mnemonic, 2019-03-20:
Check [this commit](https://github.com/nunofgs/docker-octoprint/commit/e9ef24193b440d0d4313126b6c8088769e09090d). This solution seems to work fine.

ViolentCrumble, 2019-05-20:
> I've had the same problem and found a really hacky solution for it. Seems to work for me.
> The trick was similar to what @stickystyle proposed.
>
> Well, -t /home/octoprint/.octoprint/plugins didn't [...]

could you provide any clearer instructions on the commands required to do this? having some trouble.

ViolentCrumble, 2019-05-20:
I think i am finally getting closer,  This is my docker-compose file, basically i am running 2instances of octoprint and everything boots up fine. I install a plugin and cannot for the life of me workout how to restart octoprint.. I use docker restart <containerID> and it restarts and the plugin shows up.

However when i use docker-compose down and then use docker-compose up again the plugin is gone.

Is this the correct way to mount the venv directory? This is running on ubuntu 18.04. I have tried using ./Public/venv1 and also tried using ./venv. I have tried /home/octoprint/venv and /opt/octoprint/venv

nothing seems to persist after a docker-compose down. For now it is usable but useless to me, if the power goes out or something else i will have to configure everything all over again.

Any help would be appreciated. thanks!

```
version: '3.2'

volumes:
  octoprint-docker:
  octoprint-docker2:
  ffmpeg1:
  ffmpeg2:
services:
  octoprint1:
    image: octoprint/octoprint
    ports:
      - 5000:5000
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    volumes:
      - octoprint-docker:/home/octoprint/.octoprint
      - ffmpeg1:/home/ffmpeg
      - ./venv1:/home/octoprint/venv
    restart: always

  streamer1:
    build: .
    ports:
      - 8911:8080
    devices:
      - /dev/video1:/dev/video0
    environment:
      - MJPG_RES=640x480
      - MJPG_FPS=25
      - MJPG_USER=admin
      - MJPG_PASS=adm1n
    restart: always

  octoprint2:
    image: octoprint/octoprint
    ports:
      - 5001:5000
    devices:
      - /dev/ttyUSB1:/dev/ttyUSB1
    volumes:
      - octoprint-docker2:/home/octoprint/.octoprint
      - ffmpeg2:/home/ffmpeg
      - ./venv2:/home/octoprint/venv
    restart: always

  streamer2:
    build: .
    ports:
      - 8912:8080
    devices:
      - /dev/video0:/dev/video0
    environment:
      - MJPG_RES=1280x720
      - MJPG_FPS=60
      - MJPG_USER=admin
      - MJPG_PASS=adm1n
    restart: always
```

johny-mnemonic, 2019-05-20:
@ViolentCrumble I already posted working solution in the comment above.

ViolentCrumble, 2019-05-20:
> 
> 
> @ViolentCrumble I already posted working solution in the comment above.

Hi, I tried that first and it did not work. I use docker-compose and i added in the line into my DockerFile but yeah couldn't work out how to make it work. It was just some environment variables right? Not sure why it didn't work but if you have any further info let me know.

Never mind tho as i ended up getting it to work the hacky way.

johny-mnemonic, 2019-05-20:
> @ViolentCrumble 
> Never mind tho as i ended up getting it to work the hacky way.
Well, I always prefer nice and clean solution, if it's available ;-) 

I am using that solution via docker-compose for few months already and I migrated my Octoprint container between three docker machines and two filesystems (providing the docker volume). So far it works flawlessly.
Those environment variables are nothing specific to Octoprint. Those are Python and Pip options, so they should work with any Octoprint install without an issue. 
The question is, whether you are using persistent docker volume correctly. I would highly recommend using the container from that linked repo, to save the hacking time ;-)

ViolentCrumble, 2019-05-20:
sure if i can work out how to use it i will. I have my named volume working now. 

I use docker-compose and i tried adding it to my docker-compose file but it did not seem to work, i tried adding it to my dockerfile and it didn't work.. I tried adding it like this in my docker-compose file and no luck.

```
environment:
    - DEBUG=1
```
but that might have been because back then my volume was not working correctly.

johny-mnemonic, 2019-05-21:
I would bet that the volume was the main issue, but seems to me, that your format of environment variables is not OK.
[Here](https://gist.github.com/johny-mnemonic/58c48ea7128747ccf2b03cb15a0aed67) is my compose file for inspiration.
Do not use it 1:1, as it started as docker swarm stack file and I was lazy to clean it ;-)

TheFitzZZ, 2019-05-22:
Can confirm that the nunofgs/octoprint version works flawlessly for me and made me a happy octoprint user again :-) Thanks @johny-mnemonic for pointing it out!

badsmoke, 2020-03-22:
i need help with the implementation of this docker image.

PIP_USER true
PYTHONUSERBASE /home/octoprint/.octoprint/plugins

unfortunately do not bring the desired effect.

is there anyone here who knows about pip?

PIP_TARGET=/home/octoprint/.octoprint/plugins

is also not a perfect solution as he then packed all the other flask,websocket.... also shows in the pluginmanage

LongLiveCHIEF, 2020-03-25:
Python apps, especially those that cause `pip` commands to run, aren't ideal for dockerization.  I'm working on a few things over the next week that should be able to support this.

The idiomatic way to do this would be to use the new images as a base in your own `Dockerfiles`, and then running a `pip install` for any plugins you want to install as a `RUN pip install` instruction in your own dockerfile, and then running a new `docker build` when new docker images are released.

Remember, docker is not an operating system, it's a process wrapper. It's not _meant_ to do this, and if you _do_ do this, then when you switch to new images of octoprint, things will likely break.

I'll be working on more documentation and configuration over the next week or so, and I also have a technique with PATH and virutalenv to test out that should let you do the things being asked of in this thread (even though again I'll stress that you do so at your own risk).

Some of the permissions issues should have already been fixed with the images deployed as a result of #19 2 days ago, and I'd appreciate any updates you can provide with your own testing of those images (which were pushed to the official ocotoprint/octoprint tag on docker hub).

We're now publishing both python 2 and python 3 versions of from both master and latest stable release (1.4.0 at the time of this comment).  Multi-arch images will also be supported, and I'm planning on deploying those within the next 2 or 3 days.

Check them out on hub.docker.com/r/octoprint/octoprint

I also added a new [CONTRIBUTING.md](/OctoPrint/docker/blob/master/CONTRIBUTING.md) set of instructions and documents that may be of help if you want to build your own images for testing.

In the meantime, you can catch me on the #octoprint channel on freenode, or join my HackerHappyHour discord server and ask questions there. This discord invite link will take you directly to the raspberr-pi channel on that server: https://discord.gg/hGwvAPp.

LongLiveCHIEF, 2020-03-30:
closing in favor of #22

closed 2020-03-30 by LongLiveCHIEF

RyanBalfanz, 2020-04-28:
Here is another option: https://github.com/OctoPrint/docker/issues/22#issuecomment-620844676