OctoPrint/octoprint-docker #12: Stop using latest tag for betas/dev/rc version

reloxx13, 2018-11-14:
Heya,

pls dont use the docker latest tag for dev versions.

add a second tag like dev/beta/deveoper which points to beta versions and update latest tag only if its a stable release.

LongLiveCHIEF, 2020-03-30:
Agreed. moving forward, latest will point to the latest stable release of Octoprint. So for now, that would mean `latest` and `1.4.0` tags would point to the same digests.

I'll be making this update later today.

LongLiveCHIEF, 2020-04-04:
#35 switched the `latest` tag to be equal to the latest stable release, but I will need to validate that my version.sh script will return the latest stable, and not any RC's that are released.

gaetancollaud, 2020-04-05:
Since #35 is merge I will close this one.

closed 2020-04-05 by gaetancollaud