OctoPrint/octoprint-docker #17: new docker image

badsmoke, 2020-03-07:
I have build a newer docker image for octoprint.

octoprint verison 1.40
python version 3.8

https://hub.docker.com/repository/docker/badsmoke/octoprint

mfg

gaetancollaud, 2020-03-09:
My 3D printer is taking dust and I'm not using octoprint that much anymore.

Are you a heavy user ? Would you like write permission of this repos ? So you can put you work here.

badsmoke, 2020-03-09:
yes of course, I will take care of it

gaetancollaud, 2020-03-09:
@foosel I don't have the right. Can you please add @badsmoke  as maintainer of this repo ?

foosel, 2020-03-09:
@gaetancollaud actually you *should* have that right but I've also just taken care of that

foosel, 2020-03-09:
Side note: not so sure it's a good idea to switch the docker image to python 3 yet - the plugin ecosystems really ain't there yet

badsmoke, 2020-03-10:
all right, this will be a simple two builds, python2 and python3

LongLiveCHIEF, 2020-03-21:
I'd also be willing to stand by as a maintainer, just so you don't have a single maintainer.

gaetancollaud, 2020-03-22:
@LongLiveCHIEF I added you

@foosel you're right I just didn't found it the first time.

I'm closing this issue.

gaetancollaud, 2020-03-22:
@LongLiveCHIEF @badsmoke Once you're done, can you please have a look at the issues and close the one that are not relevant anymore:  https://github.com/OctoPrint/docker/issues

Thanks in advance for taking care of this

closed 2020-03-22 by gaetancollaud

LongLiveCHIEF, 2020-03-22:
Thanks @gaetancollaud 

@foosel @badsmoke I am working on a PR for a multi-stage build [https://github.com/LongLiveCHIEF/octoprint-docker/blob/multi-stage-build/Dockerfile] see notes about the steps I'm taking in the [commit history](https://github.com/LongLiveCHIEF/octoprint-docker/commits/multi-stage-build/Dockerfile), but I have a question about the build technique (wheel vs virtualenv). 

Are you both on the freenode channel, or is there a discord group or some other way that maintainers collaborate?

meteozond, 2020-03-23:
Got error renaming /home/octoprint/.octoprint during config backup import process with latest image.

gaetancollaud, 2020-03-23:
> Got error renaming /home/octoprint/.octoprint during config backup import process with latest image.

Could you please open a new ticket so we can better track issues please.

LongLiveCHIEF, 2020-03-23:
@gaetancollaud can you double check that you added me?  I want to add a milestone for multi-arch build/publish, and track issues towards it.

I can't access any of the capabilities a member should have (create projects/milestones/labels/releases/etc...).

LongLiveCHIEF, 2020-03-23:
also, can you add me to the octoprint organization on docker hub?  My username there is `longlivechief`

gaetancollaud, 2020-03-23:
@foosel this time it's for real, I'm not admin of the docker hub organization ;) 

can you add @LongLiveCHIEF  to https://hub.docker.com/orgs/octoprint please ?

foosel, 2020-03-23:
Should be taken care of

LongLiveCHIEF, 2020-03-23:
@foosel it was actually a python build question, but I got it figured out.

LongLiveCHIEF, 2020-03-23:
@foosel @gaetancollaud I'm also still missing all permissions for _this_ repository. :smiley:

docker hub permissions are all set though! :heart:

foosel, 2020-03-23:
@LongLiveCHIEF according to what I see, you should have gotten an invite to this project

LongLiveCHIEF, 2020-03-23:
I went to the invitations api and found it!  The email must have gotten blocked by spam filter or something.

:man_shrugging: 

Thanks!