OctoPrint/octoprint-docker #21: fix deploy stage of travis.yml

LongLiveCHIEF, 2020-03-26:
Deploy currently doesn't work because of the disconnect between stages. (Oh Travis how I hate thee).

If you have a `deploy` travis by default runs it on every script (oh travis how i hate thee),  which means we need to figure out a way to do CI with merge requests that won't overwrite the main tags for octoprint/octoprint until it's in master.

Inspirations for the solution can be take from both [node's approach with docker](https://github.com/nodejs/docker-node) and travis, as well as the [moby/buildkit](https://github.com/moby/buildkit) project's approach.

One thing to note though, is that both of the travis configs for those projects seem to use deprecated techniques for deploy, and the new syntax for travis deploy's actually makes it _harder_ (oh travis how I hate thee)

gaetancollaud, 2020-03-26:
I'm not against moving to github action or any other CI tool if you want ! I just try them yesterday for a project and it seems to work fine (although there is some limitations).

Not sure if it's fully free though.

LongLiveCHIEF, 2020-03-26:
I'll most likely stick with Travis, since contributors are used to it, and likely to have accounts that their forks will run their builds on.  But who knows. This is the first thing I'm working on with the current project, so I'll add updates as I have them.

gaetancollaud, 2020-03-26:
I was the main contributor. And I used travis because it was a viable solution at the time.

If you feel more confortable with something else I'm open.

LongLiveCHIEF, 2020-03-27:
I am going to look into github actions for this. I really like the way the node project uses this to publish multi-arch images. I may even use one of my cloud providers for an actions runner depending on performance and cost.

LongLiveCHIEF, 2020-03-28:
Looks like with my current permissions, I can't access the `Settings > Actions` to be able to add Self-hosted runners.

LongLiveCHIEF, 2020-03-28:
for now it's not a big deal. The only thing it really impacts is my automated testing strategy for physical usb/serial/gpio devices.

gaetancollaud, 2020-03-28:
I added a self-hosted runner. It runs on my server. I will monitor the resources needed.

I enabled docker build normally. But this is a x64 machine, I don't think it will be able to build ARM images.

gaetancollaud, 2020-03-28:
I just created a dummy workflow to build the image. It seems to work well.

Let me know if you want to add a runners yourself, I will give send you the secret.

LongLiveCHIEF, 2020-03-28:
Actually, I got it working for multi-arch images on the standard runners.
And on Travis also.

Right now I'm just tweaking things in the build matrix so it only builds
the tagged variants (1.4.0) when there are changes, and otherwise just
moved the tags for latest variants.

LongLiveCHIEF, 2020-03-28:
I just don't commit directly to master :smirk:  I know actions workflows will do that by default, which is poor design if you ask me.  I'm doing all of my workflow testing in my fork, and I've updated the configuration so it will allow a user to run it in their own space without tagging it octoprint/octoprint.

gaetancollaud, 2020-03-28:
No worries I just build, I didn't push anything

LongLiveCHIEF, 2020-03-29:
@gaetancollaud @badsmoke I just pushed a [multi-arch-build](https://github.com/OctoPrint/docker/tree/multi-arch-build) branch that [should build travis in multi-arch](https://travis-ci.org/github/OctoPrint/docker/builds/668378754). I've made it so users can fork and run it themselves merely by adding some travis keys as overrides (see [CONTRIBUTING.md](https://github.com/OctoPrint/docker/blob/multi-arch-build/CONTRIBUTING.md#building) changes)  This exact commit is what I used in my own fork that wound up building the multi-arch [longlivechief/octoprint](https://hub.docker.com/r/longlivechief/octoprint/tags) images. 

This also includes a compile of ffmpeg compatible with the architecture of the resulting image via the ffmpeg build stage. I wanted to check in with you guys on my progress thus far to see if you had any questions or suggestions.

My next step is to do the same for github actions, and compare the two.

I have a feeling github actions will be the one to go with, especially for its use with cache to speed builds and deploy multi-arch manifests.

Right now the biggest thing that remains unsolved is the best approach to building the different tags of octoprint.  Should we trigger a nightly or weekly build that will pull master and the latest stable tag from foosel/octoprint, and publish those?  I'd appreciate any thoughts you guys have on that.

/cc @foosel @markwal @BillyBlaze @jubaleth

gaetancollaud, 2020-03-29:
Thanks for all your work.

Up until now I always trigged build manually every time I receive a newsletter from @foosel saying that a new version is here.

Regarding the trigger of the build. I would suggest to sync with the official octoprint release. Meaning that when a new version is built (like '1.4.0` for example), we should have a new tag with this version `octoprint/octoprint:1.4.0`. `:latest` should always point to the latest stable version and not to the master branch in my opinion.

We can always have nightly build or even a build every commit on the main repo (no idea how to make trigger works between repos). We could call it this tag `:dev`

I'm new to multi-arch and docker but I think we could have one tag with multiple architecture. So we don't even need to put the name of the arch in the tag as prefix or suffix.

LongLiveCHIEF, 2020-03-29:
> I'm new to multi-arch and docker but I think we could have one tag with multiple architecture. So we don't even need to put the name of the arch in the tag as prefix or suffix.

_Every_ tag is multi-arch.  Right now, this branch is creating 2 tags. both build from `foosel/OctoPrint` master. The one that uses python 2.7 is tagged `latest` and available in all architectures we support, and then there's another tag of `python3` that is also available in all architectures we support.

The idea behind multi-arch images is that you just focus on the _tags_ you want, and _all_ of them are multarch aware.  The point of this is that as a user you can pull the tag you want from whatever platform you want, and you're guaranteed to get an image that will run on your system.

This means that if you pull `octoprint/octoprint:latest` or `octoprint/octorpint:1.4.0` or any mix thereof, it will run regardless if you are on a 32bit raspberry pi, 64 bit arm, 64 bit amd, etc...  You don't have to figure out what your arch is and pull a corresponding tag.

All the official docker images work like this. You can pull any python tag from any host and not worry about looking for your architecture specific tag.

LongLiveCHIEF, 2020-03-29:
The travis build i linked to above had two build stages,built 4 images, and created 2 manifests with 2 digests each.

LongLiveCHIEF, 2020-03-29:
confirmed, github actions with remote cache makes builds extremely fast. I can build all 4 images in under a minute: https://github.com/LongLiveCHIEF/octoprint-docker/actions.

I don't know why, but I've lost my ability to manage actions for this repository, even though I had it just the other day.  I'm just about ready to submit the PR for this, but can't do so until I can manage secrets.

If I'm going to be a maintainer, I'll need access.  @foosel @markwal @BillyBlaze @jubaleth could one of you go into the OctoPrint organization on github, and create a docker-maintainers team, and give admin access to just this repo for members of that team, and then add myself, @badsmoke and @gaetancollaud to it?

That teams feature will give us the access we need for this repo without granting us any access to the rest of the Octoprint org repositories.

Thanks!

LongLiveCHIEF, 2020-03-29:
@gaetancollaud I've officially fallen in love with github actions.  This was actually my first chance to use them.  I was going to build my own buildx action for multi-arch builds, but then discovered https://github.com/crazy-max/ghaction-docker-buildx. It literally allowed me to replicate everything I'd done in travis over 2 days in about 20 minutes using actions.

closed 2020-03-30 by LongLiveCHIEF

foosel, 2020-03-30:
>  @foosel @markwal @BillyBlaze @jubaleth could one of you go into the OctoPrint organization on github, and create a docker-maintainers team, and give admin access to just this repo for members of that team, and then add myself, @badsmoke and @gaetancollaud to it?

Done. No idea why you lost action access, but I hope this is solved now.

LongLiveCHIEF, 2020-03-30:
@foosel thanks! :heart: It's been a wild week, so probably just a PEBKAC error on my part. (I think I was confusing being able to edit secrets on the travis org with secrets on this repo) 

Everything looks good now, and we're cooking with gas!

Thanks for all you've done on this, and thanks for letting me be a part of it in this small way. I'm glad i can lend my expertise to a project I've been a consumer of for years!

Take a look at the [Stable Automated Docker Releases](https://github.com/OctoPrint/docker/projects/1#card-35221562) and shout at me on IRC if you have any questions.

PR #27 (2020-03-30): Multi arch build
Switches from travis-ci to github actions. Will require secrets to be added in settings > secrets for `DOCKER_USERNAME` and  `DOCKER_PASSWORD`.

fixes #21
fixes #8 #18