OctoPrint/octoprint-docker #23: Permission denied for home.config.yaml

scubieman, 2020-03-26:
Not sure how to fix this error
Trying to follow this video also   https://www.youtube.com/watch?v=Y-2o0gyGB4c
>   
version: '3.7'

services:
  octoprint:
    image: octoprint/octoprint
    ports:
      - 5000:5000
    # devices:
    #  - /dev/ttyACM0:/dev/ttyACM0
    volumes:
     - ./config:/home/octoprint/.octoprint

Here is my docker-compose.yaml

>   GNU nano 4.5                                        docker-compose.yaml                                                   

version: '3.7'

volumes:
  octoprint-docker:

services:
  octoprint:
    image: octoprint/octoprint
    ports:
      - 5001:5001
    # devices:
    #  - /dev/ttyACM0:/dev/ttyACM0
    volumes:
     - octoprint-docker:/home/octoprint/.octoprint

LongLiveCHIEF, 2020-03-26:
Watch the whole video :smiley: ... specifically at [2:00](https://youtu.be/Y-2o0gyGB4c?t=120) :wink:

LongLiveCHIEF, 2020-03-26:
You'll also want to make sure the port inside the container is `5000`.

scubieman, 2020-03-26:
Couple things, I tried to change to same as the 2:00 mark.

Also I tried port 5000. Just running two different dockers of unraid Thats why i have 5001.

Also my yaml matches the 2 minute mark

LongLiveCHIEF, 2020-03-26:
Would you be willing to reach out to me on the #octoprint channel on freenode, or join my HackerHappyHour discord server and live-debug this one with me? This discord invite link will take you directly to the raspberry-pi channel on that server: https://discord.gg/hGwvAPp.

The only thing wrong I see with your yaml pasted above is formatting, but that could just be a result of using a blockquote instead of code insertion backticks.

when you type `docker-compose config` do you get an error, or does it spit out a copy of your resolved configuration?

Also, when you type `groups` can you confirm that the following groups all show up `docker` `plugdev`, `dialout`, `tty`

LongLiveCHIEF, 2020-03-26:
Looks like I forgot to add back in user membership to `tty` and `dialout` groups for access to `/dev/ttyUSB*` and `/dev/ttyACM*`.

PR #24 (2020-03-26): Add octoprint user to groups
fixes #23
commit bb81ecd8 (2020-03-26): updated example docker-compose file

closes #23
closed 2020-03-26 by LongLiveCHIEF