OctoPrint/octoprint-docker #29: README updates for tag list

LongLiveCHIEF, 2020-03-30:
Update README to include a list of docker tags, similar to that you would find on all the official images on docker hub.

Each tag should link back to the respective Dockerfile (at the moment they're all built from the same one)

LongLiveCHIEF, 2020-07-31:
Here's the tagging convention we've decided to go with:

```
latest, X.Y.Z, X.Y, X, latest-python3, X.Y.Z-python3, X.Y-python3 X-python3
latest-camera, X.Y.Z-camera, X.Y-camera, X-camera, X.Y.Z-python3-camera, X.Y-python3-camera, latest-python3-camera
latest-python2, x-python2, X.Y-python2, X.Y.Z-python2
latest-python2-camera, x-python2-camera, X.Y-python2-camera, X.Y.Z-python2-camera
```

See relevant discussion on discord: https://discordapp.com/channels/704958479194128507/712573137325391973/736600275397247016

Mearman, 2020-07-31:
Are the `X.Y` versions going to be the latest patch version for that minor version?

LongLiveCHIEF, 2020-08-01:
@Mearman that's correct.  It'll follow the conventions of most official images. When a new patch is released, each row listed above will all point to the same image digest.

commit 28e8a8d6 (2020-08-23): add tag pattern to README

closes #29
closed 2020-08-23 by LongLiveCHIEF