OctoPrint/octoprint-docker #37: Add gpio access and control and docs

LongLiveCHIEF, 2020-04-04:
On a test of the OctoPrint PSU control plugin via docker, I believe it is necessary to enable and document a few things related to using GPIO for our users.

The following 2 docker container settings are required for gpio dependent plugins to work:
- `cap-add SYS_RAWIO`
- `--device /dev/mem`

Additionally, the gpio device tree needs to be made writeable, and any pins that are controlled by octoprint plugins need to have corresponding INPUT/OUTPUT modes set, and the pin number registered in the /sys/class/gpio device tree.

This can be done manually with the `gpio` ubuntu package available from the `wiringpi` package in the default ubuntu registry.

I'm not sure if these plugins use this program, as I can't remember if I had to set things up manually the first time I installed PSU Control on a normal Octopi Raspbian image.

Tasks:
- [ ] determine best practice for adding gpio channel manipulation so that the users have easiest possible experience when using plugins that utilize gpio controls
- [ ] add any updates needed to Dockerfile or entrypoint script based on the determined best practice
- [ ] determine if gpio plugins will automatically control pin mode if the appropriate system devices and packages are present, or if these need to be set manually

Documentation:
- [ ] add docs about any necessary host commands/configs
- [ ] docker-compose settings for `cap_add` and `--device /dev/mem`
- [ ] any downstream Dockerfile changes needed to support gpio control if turns out that's the best practice

LongLiveCHIEF, 2020-04-04:
The more native way of accessing things would be to use the sysfs interface.  It's possible that's how PSU Control and other plugins work (via whatever python packages they use), and we merely need to map in `/sys/class/gpio` in addition to using `cap-add SYS_RAWIO`.

LongLiveCHIEF, 2020-04-04:
looks like at least PSU Control imports and uses [RPI.GPIO](https://sourceforge.net/projects/raspberry-gpio-python/) to do what it does.

LongLiveCHIEF, 2020-04-04:
Looks like no matter what we do, even adding udev rules to hosts and giving the user godlike permissions, the final result is that the specific library i'm using to test (PSU Control) uses RPI.GPIO module. Even with all permissions between host and container fixed, it will throw an error:

```
octoprint.plugins.psucontrol - ERROR - Not running on a RPi!
```

@OctoPrint/octoprint-maintainers something to consider here... It might be a best practice moving forward if plugins avoid the use of libraries that make an assumption about the OS they are run on. For example we could recommend to use https://github.com/vitiral/gpio `import gpio` for OctoPrint which aims to use the same api as RPI.GPIO, but uses the native sysfs interface on all linux kernels, instead of packages that only exist on raspbian.

Not sure what next steps to go for this one just yet, as there is still some of the documentation pieces that are good to cover. @OctoPrint/docker-maintainers any thoughts here?

---

`/cc` @kantlivelong I wanted to open an issue in the PSU Control plugin github repo, but issues are turned off.  Is that repo migrating to the OctoPrint org and that's just temporary, or should issues with that plugin be opened in OctoPrint repo? I believe we can add a few lines of code to the existing code so it will run smoothly even in a container even without changing libraries, and wanted to get the issue on paper in case someone else has time to work on it. I won't be able to get to it for a while.

I also saw an opportunity to improve some guidance in the wiki for your plugin. The troubleshooting question "can't access /dev/mem"  links to a SO post has a technically correct answer, but only under a small number of conditions.  The best fix for that problem is to create a udev rule for gpio devices assigning them the group and controls required to survive reboots. Let me know if that sounds ok to you and I'll drop the changes in gist that you can copy.

kantlivelong, 2020-04-04:
@LongLiveCHIEF I turned it off as people were abusing it as support instead of a tracker. I was actually looking at replacing RPi.GPIO with generic GPIO support using [python-periphery](https://github.com/vsergeev/python-periphery). Have even thought of creating a separate GPIO plugin with exposed helpers to be used by other plugins. I believe the biggest problem would be compatibility with other plugins that continue to use RPi.GPIO.

kantlivelong, 2020-04-04:
PS: I'm generally in #octoprint on Freenode if you want to discuss it. :smile:

gaetancollaud, 2020-04-05:
Regarding the RPI detection and the error `octoprint.plugins.psucontrol - ERROR - Not running on a RPi!` I think that the plugin should detect that we are on docker on a RPI. We could of course ease this detection by providing some environments variables or create a file at a specified location.

For the GPIO I'm not an expert but can't we just mount devices and volumes ? I don't think we need to do anything specific (except maybe document it).

LongLiveCHIEF, 2020-04-05:
@gaetancollaud It's not the plugin that throws this error, it's the `RPI.GPIO` module it uses internally.

https://sourceforge.net/p/raspberry-gpio-python/code/ci/default/tree/source/py_gpio.c

The changelog for that module has some notes about updating detection for aarch64, and also some open issues about this.

LongLiveCHIEF, 2020-04-05:
so yeah... looks like it's not specific to docker, which is good to know.

eduncan911, 2020-08-04:
I'll give another scenario... Raspberry clones, specifically Intel based ones.

I have a large collection of UP Boards (Intel Atom, Celeron, and Pentium x64 powered).  They are direct RPi formats, using the same 40-pin GPIO.

I've done a lot of development on RPi hats for these boards, and it basically comes down to:

- The drivers and software must be buildable under x64 Arch.  A lot of the HATs release proprietary blobs built only for ARM arch, preventing their use on x64.
- Run as root :(

I think it's because of the GPIO hacks to the x64 Linux kernel that requires root to access to the men space.

To sum up, just an FYI to not prevent things running as root as there are use cases that we may have to.

shift, 2020-09-02:
FYI http://wiringpi.com/wiringpi-deprecated/

LongLiveCHIEF, 2020-09-02:
yeah, doesn't surprise me. I already wrote some stuff that uses the new character device libraries instead of relying on wiringpi.  Forgot where I put it though.

@jneilliii and I were tossing around some ideas using python-periphery (utilizes the native character device interface of recent kernels), but at the time octoprint was still python v2, and arm64 hadn't really become officially supported by Ubuntu yet, meaning wiringpi worked for most things, so it wasn't too much of a problem.

Might be about time to take a closer look now though.

jneilliii, 2020-09-02:
The python 2 will be dropped in the next octopi release 0.18, whenever that is. The nightlies are already building python 3 only images.

Oxmaster, 2020-09-24:
So does the GPIO work with docker? I'm using IOTstack portainer with octoprint and can't use octoprint-enclosure plugin at all, it says that it installed but after restart the plugin is not running and I can install it again. I'm a total noob when it comes to linux, docker and this kind of things.

```
Installing plugin "OctoPrint-Enclosure" from https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip...
/usr/local/bin/python -m pip --disable-pip-version-check install file:///tmp/tmp9s3uflao/OctoPrint-Enclosure-master.zip --no-cache-dir
Processing /tmp/tmp9s3uflao/OctoPrint-Enclosure-master.zip
Requirement already satisfied: OctoPrint in /usr/local/lib/python3.8/site-packages (from OctoPrint-Enclosure==4.13.1) (1.4.2)
Requirement already satisfied: RPi.GPIO>=0.6.5 in /root/.octoprint/plugins/lib/python3.8/site-packages (from OctoPrint-Enclosure==4.13.1) (0.7.0)
Requirement already satisfied: requests>=2.7 in /usr/local/lib/python3.8/site-packages (from OctoPrint-Enclosure==4.13.1) (2.24.0)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.1.0)
Requirement already satisfied: netifaces<1,>=0.10.9 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (0.10.9)
Requirement already satisfied: flask<2,>=1.1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.1.2)
Requirement already satisfied: feedparser<6,>=5.2.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (5.2.1)
Requirement already satisfied: tornado==5.1.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (5.1.1)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (3.4)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.5.0.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.12.1)
Requirement already satisfied: emoji<1,>=0.5.4 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (0.6.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (2.8.5)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.06.22 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (2020.6.22)
Requirement already satisfied: regex!=2018.11.6 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (2020.7.14)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (2.11.2)
Requirement already satisfied: cachelib<1,>=0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (0.1.1)
Requirement already satisfied: pylru<2,>=1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.2.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.0.0)
Requirement already satisfied: psutil<6,>=5.7 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (5.7.2)
Requirement already satisfied: frozendict<2,>=1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.2)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (5.3.1)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.1.1)
Requirement already satisfied: watchdog<1,>=0.10.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (0.10.3)
Requirement already satisfied: Click<8,>=7.1.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (7.1.2)
Requirement already satisfied: future<1,>=0.18.2 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (0.18.2)
Requirement already satisfied: filetype<2,>=1.0.7 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.0.7)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (0.17.3)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (2.0)
Requirement already satisfied: markdown<3.2,>=3.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (3.1.1)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (1.0.1)
Requirement already satisfied: rsa==4.0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (4.0)
Requirement already satisfied: sarge==0.1.5post0 in /usr/local/lib/python3.8/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.1) (0.1.5.post0)
[... cut ...]
Created wheel for OctoPrint-Enclosure: filename=OctoPrint_Enclosure-4.13.1-py3-none-any.whl size=67528 sha256=5cd5ed7449b5dabbd5d7f1aeecbc8eaf2967031183f31cdc4c719bf9aa22eb5d
Stored in directory: /tmp/pip-ephem-wheel-cache-4xmxjjpn/wheels/01/8a/7d/625fe94eb1b75f39b89c7dcdc86a1ae1dd35a60a057ac15e2d
Successfully built OctoPrint-Enclosure
Installing collected packages: OctoPrint-Enclosure
Attempting uninstall: OctoPrint-Enclosure
Found existing installation: OctoPrint-Enclosure 4.13.1
Uninstalling OctoPrint-Enclosure-4.13.1:
Successfully uninstalled OctoPrint-Enclosure-4.13.1
Successfully installed OctoPrint-Enclosure-4.13.1
Done!
```

albertopc, 2020-12-24:
Hi all,
my relay is connected to gpio18 and it works on the host raspi system.
And I managed to make PSU Control works for me adding this to docker-compose:

    volumes:
      - /sys/class/gpio/gpio18:/sys/class/gpio/gpio18

and this commands under PSU Control plugin:
- On System Command: echo "out" > /sys/class/gpio/gpio18/direction
- Off System Command: echo "in" > /sys/class/gpio/gpio18/direction

LongLiveCHIEF, 2020-12-24:
@albertopc that is the deprecated interface we're talking about.

LL1r1k, 2021-02-27:
@albertopc, what exactly needs to install PSU Control plugin? I get an error:  
```
ERROR: Failed building wheel for RPi.GPIO
```

LongLiveCHIEF, 2021-02-27:
added `gcc` back in with #178, however you should know that RPi.GPIO is deprecated and no longer maintained.

morganchristiansson, 2021-03-07:
I just got this working thanks for advise in this thread.

@albertopc Thanks for advise! I see you are setting mode=input to turn off relay. I was struggling with the same issue and couldn't get it to work normally. Turns out it should be powered by 3.3v not 5v. And may damage your Pi according to comments at https://raspberrypi.stackexchange.com/questions/59668/5v-relay-will-not-close-unless-using-gpio-cleanup

With this issue resolved I got the following system commands working:
- On: `echo 1 >/sys/class/gpio/gpio4/value`
- Off: `echo 0 >/sys/class/gpio/gpio4/value`
- Sensing: `grep 1 /sys/class/gpio/gpio4/value`

Also running `echo 4 > /sys/class/gpio/export` from /etc/rc.local on host boot.

LongLiveCHIEF, 2021-03-07:
@morganchristiansson This kernel interface (sysfs) is deprecated, and tools that support it RPi.GPIO for example, have also dropped support or discontinued development. 

The new interface for gpio manipulation is the chardev interface.

morganchristiansson, 2021-03-08:
Oh yeah it says in gpio/sysfs.txt

> THIS ABI IS DEPRECATED, THE ABI DOCUMENTATION HAS BEEN MOVED TO
> Documentation/ABI/obsolete/sysfs-gpio AND NEW USERSPACE CONSUMERS
> ARE SUPPOSED TO USE THE CHARACTER DEVICE ABI. THIS OLD SYSFS ABI WILL
> NOT BE DEVELOPED (NO NEW FEATURES), IT WILL JUST BE MAINTAINED.

It does seem to say it will be maintained and included in future kernels but no new features.... But yeah. Deprecated.

There's a bunch of shell tools: raspi-gpio gpio gpioset etc that do the same thing. I wasn't sure they're in the docker image so it was nice to use just echo and grep. Hopefully PSU control plugin can just support running inside docker in future version.

jneilliii, 2021-03-08:
I know the PSU Control plugin has a version that kantlivelong has been working on that utilizes periphery instead of RPi.GPIO which may open it up to better docker compatibility.

LongLiveCHIEF, 2021-03-13:
> Btw the auto-power on/off by relay is like magic. So satisfying when it turns off 10 minutes after print. And turns on when you click print button in your slicer.

For what it's worth, I've completely decoupled power features in octoprint from the printer _and_ the octoprint host, and instead use home assistant with an esp01 to control a relay.  OctoPrint can send a mqtt message when print is finished, and that auto-switches the printer to off.  This also means I can control shutdown in case of crash of octoprint as well.

morganchristiansson, 2021-03-14:
> use home assistant with an esp01 to control a relay

I also considered this. I already have a couple Sonoff S20 plugs which have esp8266 inside [and you can flash your own mqtt firmware on them](https://esphome.io/devices/sonoff_s20.html).
And I've seen you can integrate octoprint with homeassistant.

But I'm quite enjoying my integrated 3d printing tower with 2x Ender 3 pro connected to single Pi4 with 2x octoprint docker containers. I'm going to attach ws2811b LED strip to my SKR Mini E3 2.0 boards next and get better lighting for the 2x connected cameras.

LongLiveCHIEF, 2021-03-14:
Nice. Here's mine. The photos at the bottom are the most recent. https://photos.app.goo.gl/ZiyXs526jnhR8kLz6

oerkel47, 2021-04-30:
Fyi, with the new version of PSU-Control I found that it's enough to mount /dev/gpiochip0

nothing else necessary for me.

morganchristiansson, 2021-05-01:
Thanks! I saw PSU Control 1.0 was released which should fix this. Thanks for testing it!

https://github.com/kantlivelong/OctoPrint-PSUControl/releases