OctoPrint/octoprint-docker #38: armhf: "./config" includes invalid characters for a local volume name

minnixtx, 2020-04-08:
Banana Pi M2 Zero with Armbian Bionic. Getting error message when running `$ docker run -d -v ./config:/home/octoprint/.octoprint --device /dev/ttyUSB0:/dev/ttyUSB0 -p 5000:5000 --name octoprint badsmoke/octoprint`

<details>
<summary>Error message</summary>

<pre><code>Unable to find image 'badsmoke/octoprint:latest' locally
latest: Pulling from badsmoke/octoprint
fff3167bf8c7: Pull complete 
e3a57370d71f: Pull complete 
1857b6e06e34: Pull complete 
513b7f3d59a2: Pull complete 
8a98ddf64c8c: Pull complete 
1fb6bcc7332c: Pull complete 
48070cb75b7e: Pull complete 
562ad4447872: Pull complete 
2c4cf33fc7ad: Pull complete 
6c177655d5dc: Pull complete 
5bac32922e23: Pull complete 
8dde1731c6cf: Pull complete 
0ac2c6c2dc76: Pull complete 
af429de8f6ef: Pull complete 
4b84157043c4: Pull complete 
304a48c84d30: Pull complete 
Digest: sha256:9e5e79aebd25ff19594cbecaad8f8bd66b67bc58844bfec3f30be7b0f112d1a3
Status: Downloaded newer image for badsmoke/octoprint:latest
docker: Error response from daemon: create ./config: "./config" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.
See 'docker run --help'.</code></pre>
</details>

gaetancollaud, 2020-04-08:
can you try to just do 
```
docker run -d --device /dev/ttyUSB0:/dev/ttyUSB0 -p 5000:5000 --name octoprint octoprint/octoprint
```

* without the config folder
* with the official image

minnixtx, 2020-04-08:
That worked thank you. I am up and running now.

closed 2020-04-08 by minnixtx

gaetancollaud, 2020-04-08:
Thing is, you will lose the configuration every time you destroy and recreate the container. 

I suggest you use a docker volume if you want to store the config.

minnixtx, 2020-04-08:
I see what you mean. What would be the best way to do this since the config file itself seems to be causing the error?

gaetancollaud, 2020-04-08:
You can just use a volume for the config: https://docs.docker.com/storage/volumes/