OctoPrint/octoprint-docker #39: Build fails due to MarkupSafe alpha version

markusressel, 2020-04-12:
Hey there, great project!

I just forked this repository to include some plugins in my own version of the docker image, but sadly the [build fails](https://github.com/markusressel/docker/runs/581095989?check_suite_focus=true) because of the automated version match to a recently released alpha version of MarkupSafe:

```
2020-04-12T19:03:10.5517666Z #67 137.2 Searching for MarkupSafe
2020-04-12T19:03:10.5517891Z #67 137.2 Reading https://pypi.org/simple/MarkupSafe/
2020-04-12T19:03:10.6521002Z #67 137.3 Downloading https://files.pythonhosted.org/packages/e0/bf/acc45baeb2d7333ed724c30af188640d9cb0be4b28533edfc3e2ae5aad72/MarkupSafe-2.0.0a1.tar.gz#sha256=beac28ed60c8e838301226a7a85841d0af2068eba2dcb1a58c2d32d6c05e440e
2020-04-12T19:03:10.8019546Z #67 137.4 Best match: MarkupSafe 2.0.0a1
2020-04-12T19:03:10.8022385Z #67 137.4 Processing MarkupSafe-2.0.0a1.tar.gz
2020-04-12T19:03:10.8022918Z #67 137.4 Writing /tmp/easy_install-xEbjlb/MarkupSafe-2.0.0a1/setup.cfg
2020-04-12T19:03:11.8209608Z #67 138.5 Running MarkupSafe-2.0.0a1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-xEbjlb/MarkupSafe-2.0.0a1/egg-dist-tmp-1RkEL_
2020-04-12T19:03:12.0705031Z #67 138.8 Traceback (most recent call last):
2020-04-12T19:03:12.1802607Z #67 138.8   File "setup.py", line 277, in <module>
2020-04-12T19:03:12.1802930Z #67 138.8     setup(**params())
2020-04-12T19:03:12.1804851Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/__init__.py", line 145, in setup
2020-04-12T19:03:12.1805150Z #67 138.8     return distutils.core.setup(**attrs)
2020-04-12T19:03:12.1805409Z #67 138.8   File "/usr/local/lib/python2.7/distutils/core.py", line 151, in setup
2020-04-12T19:03:12.1805811Z #67 138.8     dist.run_commands()
2020-04-12T19:03:12.1806084Z #67 138.8   File "/usr/local/lib/python2.7/distutils/dist.py", line 953, in run_commands
2020-04-12T19:03:12.1806363Z #67 138.8     self.run_command(cmd)
2020-04-12T19:03:12.1806755Z #67 138.8   File "/usr/local/lib/python2.7/distutils/dist.py", line 972, in run_command
2020-04-12T19:03:12.1807548Z #67 138.8     cmd_obj.run()
2020-04-12T19:03:12.1808891Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/install.py", line 67, in run
2020-04-12T19:03:12.1809202Z #67 138.8     self.do_egg_install()
2020-04-12T19:03:12.1810201Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/install.py", line 117, in do_egg_install
2020-04-12T19:03:12.1810499Z #67 138.8     cmd.run(show_deprecation=False)
2020-04-12T19:03:12.1811040Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 424, in run
2020-04-12T19:03:12.1811190Z #67 138.8     self.easy_install(spec, not self.no_deps)
2020-04-12T19:03:12.1811559Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 666, in easy_install
2020-04-12T19:03:12.1811710Z #67 138.8     return self.install_item(None, spec, tmpdir, deps, True)
2020-04-12T19:03:12.1812115Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 713, in install_item
2020-04-12T19:03:12.1812269Z #67 138.8     self.process_distribution(spec, dist, deps)
2020-04-12T19:03:12.1812935Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 758, in process_distribution
2020-04-12T19:03:12.1813098Z #67 138.8     [requirement], self.local_index, self.easy_install
2020-04-12T19:03:12.1813461Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 782, in resolve
2020-04-12T19:03:12.1813605Z #67 138.8     replace_conflicting=replace_conflicting
2020-04-12T19:03:12.1813965Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1065, in best_match
2020-04-12T19:03:12.1814114Z #67 138.8     return self.obtain(req, installer)
2020-04-12T19:03:12.1814464Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1077, in obtain
2020-04-12T19:03:12.1814783Z #67 138.8     return installer(requirement)
2020-04-12T19:03:12.1815187Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 685, in easy_install
2020-04-12T19:03:12.1815343Z #67 138.8     return self.install_item(spec, dist.location, tmpdir, deps)
2020-04-12T19:03:12.1815723Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 711, in install_item
2020-04-12T19:03:12.1815871Z #67 138.8     dists = self.install_eggs(spec, download, tmpdir)
2020-04-12T19:03:12.1816246Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 896, in install_eggs
2020-04-12T19:03:12.1816399Z #67 138.8     return self.build_and_install(setup_script, setup_base)
2020-04-12T19:03:12.1816786Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1164, in build_and_install
2020-04-12T19:03:12.1816944Z #67 138.8     self.run_setup(setup_script, setup_base, args)
2020-04-12T19:03:12.1817323Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1150, in run_setup
2020-04-12T19:03:12.1817470Z #67 138.8     run_setup(setup_script, args)
2020-04-12T19:03:12.1818100Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
2020-04-12T19:03:12.1818250Z #67 138.8     raise
2020-04-12T19:03:12.1818370Z #67 138.8   File "/usr/local/lib/python2.7/contextlib.py", line 35, in __exit__
2020-04-12T19:03:12.1818513Z #67 138.8     self.gen.throw(type, value, traceback)
2020-04-12T19:03:12.1818900Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
2020-04-12T19:03:12.1819038Z #67 138.8     yield
2020-04-12T19:03:12.1819166Z #67 138.8   File "/usr/local/lib/python2.7/contextlib.py", line 35, in __exit__
2020-04-12T19:03:12.1819304Z #67 138.8     self.gen.throw(type, value, traceback)
2020-04-12T19:03:12.1820065Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
2020-04-12T19:03:12.1820224Z #67 138.8     saved_exc.resume()
2020-04-12T19:03:12.1820617Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
2020-04-12T19:03:12.1821082Z #67 138.8     six.reraise(type, exc, self._tb)
2020-04-12T19:03:12.1821490Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
2020-04-12T19:03:12.1821632Z #67 138.8     yield saved
2020-04-12T19:03:12.1821981Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
2020-04-12T19:03:12.1822120Z #67 138.8     yield
2020-04-12T19:03:12.1822576Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
2020-04-12T19:03:12.1822725Z #67 138.8     _execfile(setup_script, ns)
2020-04-12T19:03:12.1823319Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
[... 5 lines cut ...]
2020-04-12T19:03:12.1825603Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/__init__.py", line 145, in setup
2020-04-12T19:03:12.1825752Z #67 138.8     return distutils.core.setup(**attrs)
2020-04-12T19:03:12.1826028Z #67 138.8   File "/usr/local/lib/python2.7/distutils/core.py", line 124, in setup
2020-04-12T19:03:12.1826237Z #67 138.8     dist.parse_config_files()
2020-04-12T19:03:12.1826909Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/_virtualenv.py", line 21, in parse_config_files
2020-04-12T19:03:12.1827502Z #67 138.8     result = old_parse_config_files(self, *args, **kwargs)
2020-04-12T19:03:12.1828423Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/dist.py", line 702, in parse_config_files
2020-04-12T19:03:12.1828579Z #67 138.8     ignore_option_errors=ignore_option_errors)
2020-04-12T19:03:12.1828960Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/config.py", line 121, in parse_configuration
2020-04-12T19:03:12.1829086Z #67 138.8     meta.parse()
2020-04-12T19:03:12.1829424Z #67 138.8   File "/opt/venv/lib/python2.7/site-packages/setuptools/config.py", line 426, in parse
2020-04-12T19:03:12.1829568Z #67 138.8     section_parser_method(section_options)
[... cut ...]
2020-04-12T19:03:12.1831564Z #67 138.8     value = getattr(module, attr_name)
2020-04-12T19:03:12.1832261Z #67 138.8 AttributeError: 'module' object has no attribute '__version__'
2020-04-12T19:03:13.3233056Z #67 ERROR: executor failed running [/bin/sh -c python setup.py install]: buildkit-runc did not terminate sucessfully
```

I did not investigate where this dependency comes from, but it would probably be a good idea to pin dependency versions per release to known and working versions (if this is not a transitive dependency).

lachyc, 2020-05-05:
I'm getting the same build fail too.

LongLiveCHIEF, 2020-05-10:
This is interesting. The docker builds automatically clone the latest stable version of Octoprint from github to run builds, and that means that upstream OctoPrint tags aren't pinned/frozen dependencies in builds.

I'm new to Python in this regards for pinning/freezing dependencies, so I'll have to chat with the other members of @OctoPrint/plugin-repository-maintainers on the best approach to this.

closed 2020-05-10 by LongLiveCHIEF

jneilliii, 2020-05-10:
I know that for my plugins the dependencies are entered in the setup.py file. I assume OctoPrint uses that somehow to run the correct pip commands, etc. I have seen issues relative to different python versions, and you can pin to versions, I just personally have never had a reported example of that being an issue with any of mine. For example, this is what I have in my Bed Level Visualizer plugin.

```
plugin_requires = ["numpy"]
```

probably should be 

```
plugin_requires = ["numpy<=1.16.4"]
```

Since that's the last version to support Python 2.7.

Hopefully that helps you guys a little.

markusressel, 2020-05-12:
It works now, thx! :+1:

PR #45 (2020-05-10): Fix buildx fail
Switched `compiler` stage to use `pip install .` instead of `python setup.py install`.  This matches the octoprint/octoprint ci build flow, and should produce consistent build success/fails as upstream, and not cause PR's to fail even when they are building stable octoprint tags.

Fixes #39