OctoPrint/octoprint-docker #40: What changes do I need to make to the compose file to interface with Klipper?

minnixtx, 2020-04-29:
Or is this possible? I don't know too much about Klipper except that it needs a virtual serial port to interface with Octoprint and I'm not sure how this would be shared outside the container. I tried https://github.com/klipperondocker/docs but was never able to access Octoprint.

gaetancollaud, 2020-04-30:
Never used Klipper but as reading the doc you have to use `/tmp/printer` as serial port. So just mount the device like in the example: https://github.com/OctoPrint/docker/blob/master/docker-compose.yml#L8-L9

```
     devices:
      - /tmp/printer: /tmp/printer
```

minnixtx, 2020-04-30:
Although I can choose /tmp/printer for my serial port within octoprint now, when I try to connect it refuses and I get this error in terminal:

<pre><code>Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: /tmp/printer SerialException: '[Errno 5] could not open port /tmp/printer: [Errno 5] Input/output error: '/tmp/printer'' @ comm.py:_openSerial:2691 (hook default)</code></pre>

minnixtx, 2020-04-30:
Ok, I think I'm getting closer with - /dev/ttyUSB0:/tmp/printer. It seems like it's trying to connect but unable to detect baudrate now. From Octoprint terminal when I hit connect:

<details>
  <summary>Octoprint terminal</summary>
  
  <pre><code>Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0xac946170, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 115200
Send: N0 M110 N0*125
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2823
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Detecting baudrate" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2823)"
Connection closed, closing down monitor</code></pre>
</details>

minnixtx, 2020-04-30:
This is really weird. When I access log `docker-compose logs -f` I get

<pre><code>octoprint_1  | Traceback (most recent call last):
octoprint_1  |   File "/opt/venv/lib/python2.7/site-packages/OctoPrint-1.4.0-py2.7.egg/octoprint/util/comm.py", line 2823, in _readline
octoprint_1  |     ret = self._serial.readline()
octoprint_1  |   File "/opt/venv/lib/python2.7/site-packages/OctoPrint-1.4.0-py2.7.egg/octoprint/util/comm.py", line 4968, in readline
octoprint_1  |     c = self.read(1)
octoprint_1  |   File "/opt/venv/lib/python2.7/site-packages/pyserial-3.4-py2.7.egg/serial/serialposix.py", line 501, in read
octoprint_1  |     'device reports readiness to read but returned no data '
octoprint_1  | SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
octoprint_1  | 2020-04-30 22:12:15,778 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
octoprint_1  | 2020-04-30 22:12:15,786 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting baudrate" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2823)"
octoprint_1  | 2020-04-30 22:12:45,243 - tornado.access - WARNING - 403 GET /api/printer (192.168.57.19) 20.98ms
octoprint_1  | 2020-04-30 22:12:45,279 - tornado.access - WARNING - 403 GET /api/printer (192.168.57.19) 24.64ms
octoprint_1  | 2020-04-30 22:12:45,316 - tornado.access - WARNING - 403 GET /api/printer (192.168.57.19) 24.29ms
octoprint_1  | 2020-04-30 22:12:45,362 - tornado.access - WARNING - 403 GET /api/printer (192.168.57.19) 24.88ms
octoprint_1  | 2020-04-30 22:12:45,395 - tornado.access - WARNING - 403 GET /api/printer (192.168.57.19) 20.40ms
octoprint_1  | 2020-04-30 22:12:45,477 - tornado.access - WARNING - 403 GET /api/printer (192.168.57.19) 21.04ms
octoprint_1  | 2020-04-30 22:12:45,666 - tornado.access - WARNING - 403 GET /api/printer (192.168.57.19) 20.64ms
octoprint_1  | 2020-04-30 22:12:45,761 - tornado.access - WARNING - 403 GET /api/job (192.168.57.19) 22.19ms
octoprint_1  | 2020-04-30 22:12:45,814 - tornado.access - WARNING - 403 GET /api/job (192.168.57.19) 21.76ms
octoprint_1  | 2020-04-30 22:12:45,869 - tornado.access - WARNING - 403 GET /api/job (192.168.57.19) 22.43ms</code></pre>

192.168.57.19 is a totally different machine on my network that has nothing to do with my printer or the device I'm running octoprint on. Why is it even being accessed?

closed 2020-05-23 by LongLiveCHIEF