OctoPrint/octoprint-docker #43: Incorrect folder mount

lachyc, 2020-05-06:
README states ` -v ./config:/home/octoprint/.octoprint` as a required mount but the Dockerfile lists `VOLUME /home/octoprint` instead. Mounting  /home/octoprint as per the Dockerfile seems to cause execution permission errors on my system. Changing to `VOLUME /home/octoprint/.octoprint` in the Dockerfile fixes these issues.

[Pull Request](https://github.com/OctoPrint/docker/pull/42)

LongLiveCHIEF, 2020-05-10:
The README needs to be updated.  Can you use `-v ./config:/home/octoprint` to mount instead?

I updated the docker-compose.yml file, but haven't yet made it around to updating the README because there are a LOT of changes, and we're not yet at the "stable" release milestone for this image.

LongLiveCHIEF, 2020-05-23:
closed #48

closed 2020-05-23 by LongLiveCHIEF