OctoPrint/octoprint-docker #44: Plugins are not persisted

M-Pixel, 2020-05-09:
If a plugin is installed, and the container is recreated, then the previously installed plugins will be lost.  Would volumizing `/opt/venv/lib/python2.7/site-packages/` be the correct way to fix this?

LongLiveCHIEF, 2020-05-10:
How are you running the container?  Are you using the `docker-compose.yml`, or a `docker run` command.

We're in the process of redoing the docs, and the new recommendation for mount will be to mount to `/home/octoprint` instead of `/home/octoprint/.octoprint`, which should persist plugins. 

related #43

LongLiveCHIEF, 2020-05-10:
closing in favor of #22

closed 2020-05-10 by LongLiveCHIEF

docb7, 2020-05-17:
Hi, although this is marked as closed, my plugins still disappear after recreating the container. And my mount is /home/octoprint. I have not been getting smarter by hopping between all those linked issues and I did not find a resolution by reading that all. But as all are marked "closed" I think I have overread something. Could someone please point me in the right direction?

LongLiveCHIEF, 2020-05-17:
The reason they are closed, is because they are all the same base problem (#22), and there is an issue open to track the progress on solving and documenting approaches to that problem. 

We don't have a stable recommendation for this problem yet, because the current architecture of OctoPrint isn't really something that works well with what docker is designed to do.

We're looking at ways to address this, but it could be that it won't be docker friendly until there are some substantial enhancements to OctoPrint itself.

In the meantime, usage of this image is highly dependent on a very solid knowledge about the workings and design on docker. This is not the solution for users who only use docker as a package deployment technique.

At the moment, this image would require a knowledge of docker equivalent with someone who is either a developer familiar with dockerizing software, or a system administrator familiar with docker or cgroups.

docb7, 2020-05-17:
Thank you so much for clearing that. As you might have guessed, I am still a docker learner, so it seems I will have to live with that issue ;-)

LongLiveCHIEF, 2020-05-17:
I'm working hard to come up the best recommendations I can for users that fit a "no knowledge of docker" usage for the existing OctoPrint code base, as well as working with the other OctoPrint developers on enhancements to make OctoPrint more docker friendly.

I just updated the README in #48 a few moments ago with some recommendations around volumes you may not have seen yet, but I still can't guarantee anything will work.

I know that a large number of octoprint docker users are using this with container based OS's and systems, and they aren't developers or sysadmins.  If at all possible, we'd like to be able to support thatt use case, and we don't expect feel that those users should be expected to have an understanding of docker fundamentals in order to use octoprint in this way.  It's just... we're a long ways off from that goal at the moment, and in the meantime, there are no guarantees.

_However_, since I'm leading this effort, I am gathering all the information I can around our users and their usage scenarios, and would love to help try to understand what your needs are and help you solve those.  That's one of the best ways we as developers have to figure out the scope of the problem, as well as the possible techniques we can use to solve them so that things are as seamless as possible for our users.

I recommend joining our discord server, and feel free to ping me on there with details about what plugins you're using, and what your setup looks like, and I will be happy to assist.

https://discord.com/invite/yA7stPp

LongLiveCHIEF, 2020-05-17:
Another thing I'll leave here, is that _my_ personal goal is to be able to run OctoPrint in a scalable cloud infrstructure such as kubernetes, and to accomplish that, I'd have to solve all the challenges that non-docker experts face anyways. 

I don't have a definitive timeline on this, but so far I've been working on it a few days each week, and @foosel and I have tossed around some ideas that may have merit for a quicker implementation that I'm currently testing.

Keep your :crossed_fingers: and continue to send as much info as you can about how you use this image (and how you _wish_ you could use this image) in the discord channel so that we don't miss anything important to our users.

Thanks!

Andoramb, 2020-07-24:
Thanks @LongLiveCHIEF !
🤞 as I've migrated to a NUC from a Pi to have everything in a buffed docker environment.
I really like the concept, but yeah, the plugins are disappearing :-/
Restorig the backup (from the Pi) solves the issue for some time though...