OctoPrint/octoprint-docker #46: Pretty much lost with octoprint docker

Shockishock, 2020-05-12:
Hey! I'm quite lost setting up my printer on octoprint.
I've got one Synology DS2018+ with Docker (with UI) and one octoprint instance runing on it.

Octoprint server is up and running and I could see any Serial Port at first so I set the octoprint instance with high privilege which mounted the ports /dev/ttyS0 -> /dev/ttyS3 and /dev/ttyUSB0

none of the port are working and I've that same error message like with the USB0 :

`Unexpected error while connecting to serial port: /dev/ttyUSB0 SerialException: '[Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'' @ comm.py:_openSerial:2691 (hook default)`

Any help would be much appreciated!

Shockishock, 2020-05-13:
I had to `sudo chmod -R 777 /dev/ttyUSB0` to solve my problem

closed 2020-05-13 by Shockishock

gaetancollaud, 2020-05-13:
I'm not sure it's wise to set `777` as permission.

Usually to access the serial port your user needs to be in the group `dialout`.