OctoPrint/octoprint-docker #47: mjpg steamer issues and thoughts

bbreton, 2020-05-16:
Hello,

I'm curious about thoughts here for mjpg steamer.  If there is interest I can do a pull request towards whichever option folks think would be best. I also include instructions for where I got stuck trying to use the info badsmoke provided.

First off thanks @badsmoke for all you did.  That said I had some issues with the instructions as is which made it a bit tricky for me coming at it not knowing mjpg at all.

The first problem I ran into is the Dockerfile didn't specify a default command.  Which meant that it took some digging into mjpg to figure out what I had to pass it. I realize a default command might not make sense given the diversity of webcams, but an example command with a working one for the rpi and a link to a helpful page about mjpg would be helpful.

Additionally the image at docker.io/badsmoke/mjpg-streamer:1.0.0 specifically is incompatible with docker on the RPI giving 'exec format error' when run.  I believe this simply means that the image doesn't support arm.  I've not done a multi-architecture build before, but this is what it needs to not require rebuilding the docker image.

If there is interest I can do docker best practices on it (build it in a multistage build etc to make it smaller, do the git pull within the Dockerfile, do multi-architecture etc.) and do a pull request.  I'm not certain if you want that in this repo though as it will be dependent on https://github.com/jacksonliam/mjpg-streamer. I could also see if they are interested in a pull request.

The other option which might be best if there is interest in having this in the same container is I could attempt to add it to the current image similar to how ffmpeg is installed perhaps with an environmental variable to enable it or not and configure the command sent to it to start rather than having it be part of the entrypoint/cmd.

In the meantime for anyone stuck on this here are some working instructions for pi-camera on a tested on a RPI4 running the up to date Raspbian Buster image:

Enable the camera with raspi-config (or google for command line method) and reboot.

```
git clone https://github.com/jacksonliam/mjpg-streamer
cd mjpg-streamer/mjpg-streamer-experimental
```
Edit the Dockerfile:
```
FROM ubuntu:18.04

RUN apt-get update && apt-get install git cmake make build-essential libjpeg-dev imagemagick subversion libv4l-dev checkinstall lib\
jpeg8-dev libv4l-0 gcc g++ -y

RUN git clone https://github.com/jacksonliam/mjpg-streamer.git

WORKDIR /mjpg-streamer/mjpg-streamer-experimental

COPY . .
RUN make USE_LIBV4L2=true clean all
RUN make install
RUN cat docker-start.sh
RUN chmod +x docker-start.sh
EXPOSE 8080/TCP
ENTRYPOINT ["/mjpg-streamer/mjpg-streamer-experimental/docker-start.sh"]
CMD ["output_http.so -w ./www", "input_uvc.so"]
```

Working docker command (the command is the same as specified as the default in the dockerfile in this case):
```
docker run -d --device /dev/video0 -e "ENV_FPS=5" -e "ENV_RESOLUTION=1280x720" -p 8080:8080 mjpg-streamer:local "output_http.so -w ./www", "input_uvc.so"
```

Other work: This doesn't include running mjpg with the "input_raspi.so" which I think would be better. I don't think this is hard, but given it is already working with input_uvc.so I went with that for now.  I'm not certain what the advantages to using input_raspi.so would be for this use case.

badsmoke, 2020-05-19:
Hello bbreton,

version 1.0.1 was built in x86,arm and arm64, always use the newest) ;-) .
I use even you raspberrys with ubuntu 18.04 and arm64, it worked fine there.

You could change it to a multistage build, but I have no problem with the current size. 

I would not like to include ffmpeg, I am more of a proponent of it: one server per image

to the point of your default settings:
This is in the docker-start.sh

```
#!/bin/sh
set -e

FRAMES=${ENV_FPS:-"15"}
RESOLUTION=${ENV_RESOLUTION:-"640x480"}
PORT=${ENV_PORT:-"8080"}
LOCATION=${ENV_LOCATION:-"/usr/local/www"}
CAMERA=${ENV_CAMERA:-"/dev/video0"}

export LD_LIBRARY_PATH="/mjpg-streamer/mjpg-streamer-experimental"
./mjpg_streamer -i "input_uvc.so -d $CAMERA -f $FRAMES -r $RESOLUTION" -o "output_http.so -p $PORT -w $LOCATION "
```

so your settings are already being executed?

LongLiveCHIEF, 2020-05-23:
In the long run, I think what we're shooting for is that you'd be able to attach other containers to the octoprint network for things like plugins/video/cura/etc.. (anything that isn't native and vanilla octoprint).

The other approach is that we create a build matrix (which the recent build refactor was actually designed to support so we could go that direction if need be), and wind up with a huge bevy of octoprint images with tags based on what features you wanted.  That's probably not ideal for the _user_.

that being said, I believe mjpg is one of the most heavily requested asks, so it wouldn't be unreasonable to add it in, and allow an ENV var to specify which video provider the user prefers.

At the moment, we're working to look at the best approach to making OctoPrint more compatible with docker in general, so some of these types of issues are being held off on so we don't wind up making massive changes to existing user implementations if the container usage changes suddenly down the road.

For now, I'll add this to the official features backlog, and it will be included as a primary test case for the octoprint docker integration designs we're considering.

/cc @foosel

LongLiveCHIEF, 2020-05-24:
So we had some good conversation about this on octoprint discord yesterday, and I discovered I had the wrong idea entirely about how vanilla OctoPrint does streaming. I thought it was there by default with ffmpeg, and I was wrong. 

Now that I've spent some time getting up to speed on how all this works, I am going to make mjpeg-streamer functionality part of the primary "stable" release target for the official octoprint/octoprint image.

Another thing that happened yesterday, is someone pointed out the [nunofgs/octoprint](https://hub.docker.com/r/) octoprint image, and I was really impressed with that image.  It has both multi-arch and mjpeg-streamer support.  

I'm going to reach out to https://github.com/nunofgs and see if he'd like to collaborate or become a member of this org and become another maintainer of this project.

I think it would be great to combine our efforts and deliver more for the octoprint community together.

I am going to use the same approach he uses for mjpeg-stream and haproxy, but I am first going to look into switching to one of the linuxserver.io base images, due to the work they've done to build supervised multi-process images. (typically docker follows that you should have one container per process).

PR #61 (2020-08-16): Add camera image variant
- uses s6 process supervisor
- uses the default octoprint image of the same version as the base image
- currently forces `/dev/video0` for webcam input 
  - will require the use of mapping your video drive to `/dev/video0` until we get the `CAMERA_DEV` var supported. Ex...( `-v /dev/video1:/dev/video0`)
- uses non-privileged port of `8888` as the haproxy frontend
- supports `arm64,amd64,armhf` architectures (same as the base image)
- only available in `slim-buster` for now (same as the base image)
- builds 3 tags automatically on releases of OctoPrint, by running the M [... cut]
closed 2020-08-16 by LongLiveCHIEF