OctoPrint/octoprint-docker #50: trigger build and tag on OctoPrint/OctoPrint master/release

LongLiveCHIEF, 2020-07-25:


LongLiveCHIEF, 2020-07-25:
Looks like there's no one perfect way to do this, but several approaches that would probably work:

1. Just set a schedule to watch for releases
2. Use a workflow in the octoprint/octprint project to send a [workflow_dispatch](https://docs.github.com/en/developers/webhooks-and-events/webhook-events-and-payloads#workflow_dispatch) or [repository_dispatch](https://docs.github.com/en/actions/reference/events-that-trigger-workflows#repository_dispatch) event on release of octoprint/octoprint, to the octoprint/docker to run specific workflows in specific branches

LongLiveCHIEF, 2020-07-25:
May also be able to use the discord bot for octoprint/octoprint to actually trigger workflows in octoprint/docker :thinking:  The key is making sure the payload is formatted correctly, and that may actually be easier to do using the discord bot than github itself (github actions really needs the ability to trigger on organization events and not just repository events)

LongLiveCHIEF, 2020-07-25:
https://github.community/t/run-action-on-event-in-another-repository/17667

eduncan911, 2020-08-04:
Yes, the second option works well.  I use it at work for several private repos.

I'd offer to help but I'm swamped with face shields for schools at the moment.

eaceaser, 2020-08-06:
I see that https://github.com/OctoPrint/OctoPrint/pull/3667 is out for review -- are you working on the `octoprint/octoprint-docker` side of things to handle the event? If not I can take a stab at it

LongLiveCHIEF, 2020-08-06:
> I see that OctoPrint/OctoPrint#3667

Yes I am.

LongLiveCHIEF, 2020-08-06:
@eaceaser sorry for the short reply earlier, I was mobile. I am working on creating a matrix of tags from the same upstream release. I've actually been live-streaming my work as I go ([part 1](https://www.youtube.com/watch?v=RJGlhPx2ECU), [part 2](https://www.youtube.com/watch?v=Mg3o1cUeGqQ), but I tend to only work on OctoPrint stuff on the weekends only, and i didn't finish in time last weekend so I'll be wrapping it up friday or saturday.

This has actually inspired me to create a new github action that I call `Tagging Strategy` ([HackerHappyHour/tagging-strategy](https://github.com/HackerHappyHour/tagging-strategy)), that will do what I'm looking to do here. 

(Basically, take a release event such as 1.4.1, and produce image tags all pointing to the same digest for `1`, `1.4`, `1.4.1` and `latest`, as well as the same tags for any variants. )

I've cloned both OctoPrint and octoprint-docker into the HackerHappyHour org that I own so I can trigger release events from that octoprint fork for testing.  

If you want to contribute or collaborate now that you're up to speed, just hop into [discord.octoprint.org](https://discord.octoprint.org) and say hello to `@CHIEFdotJS` in the `#dev-misc` channel and we can figure out how to insert you into this WIP (what fork & branch you can branch from).

LongLiveCHIEF, 2020-08-15:
fixed via #59

closed 2020-08-15 by LongLiveCHIEF