OctoPrint/octoprint-docker #52: Auto start octoprint on boot without any commands

RaptorHunterMZ, 2020-08-10:
Good day, after following your instrutions I finalily got octoprint to run, but I need to repeat the process each time a bot, is there a way for it to auto start on boot?

Mearman, 2020-08-10:
@RaptorHunterMZ what host OS are you running on?
You probably need to add `--restart unless-stopped` to your `docker run`

closed 2020-08-16 by LongLiveCHIEF

alexkander, 2024-03-18:
@RaptorHunterMZ @Mearman I am having the same issue, Do you know how resolve?

I am using this docker compose

```
version: "2.4"

services:
  octoprint:
    image: octoprint/octoprint
    container_name: octoprint
    restart: always
    ports:
      - 80:80
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
```

After boot the raspberry, the las logs it the compose are these:

```
octoprint  | 2024-03-18 00:34:43,768 - octoprint.plugins.tracking - INFO - Sent tracking event webui_load, payload: {'browser_name': 'Chrome', 'browser_version': '122.0.0.0', 'os_name': 'Windows', 'os_version': '10'}
octoprint  | [cont-finish.d] executing container finish scripts...
octoprint  | [cont-finish.d] done.
octoprint  | [s6-finish] waiting for services.
octoprint  | [s6-finish] sending all processes the TERM signal.
octoprint  | [s6-finish] sending all processes the KILL signal and exiting.
```

Do you have any suggestion?