OctoPrint/octoprint-docker #64: octoprint/octoprint image cannot connect to /dev/tty*

cab404, 2020-08-16:
OS: NixOS

Steps to reproduce: 
1. Start docker-compose file with /dev/ttyUSB0 (or any other serial port) connected
2. Try connecting to a printer

It might be because udev rules don't work inside base image you are using.

Anyways — current Dockerfile adds `octoprint` user to `dialout` group, which only works if udev does. `/dev/tty*` device files have `sudo` group, and the easiest fix is to add octoprint to `sudo` group.

LongLiveCHIEF, 2020-08-17:
This sounds like a host configuration issue.  Let me explain.

docker containers don't have udev rules because containers don't do kernel things, and udev is most definitely a kernel process.

What you describe is typically caused by the host user that is running the docker/docker-compose commands not being part of the `docker` (or whatever group the docker-daemon uses),  `dialout` and `plugdev` groups on the host machine.

When you run a container (such as the default octoprint image) as a non-root user, you either have to add privileges to your docker command, or use the `--privileged` flag and mount `/dev:/dev`.

Can you verify that your host user has the groups above, and if so, can you share your docker-compose file?

cab404, 2020-08-17:
docker-compose.yml
```
version: '3.7'

services:
  octoprint:
    image: octoprint/octoprint
    ports:
      - 5000:5000
    devices:
     - /dev/ttyUSB0:/dev/ttyUSB0
    volumes:
     - octoprint:/home/octoprint

volumes:
  octoprint:
```

groups of a user running docker-compose 
```
users wheel dialout tor networkmanager docker plugdev avahi
```

cab404, 2020-08-17:
> docker containers don't have udev rules because containers don't do kernel things, and udevadm is most definitely a kernel process.

I thought that udev doesn't work with guest groups (guest groups loose names in the host system, and mappings [are implemented by docker](https://medium.com/@mccode/understanding-how-uid-and-gid-work-in-docker-containers-c37a01d01cf))

cab404, 2020-08-17:
I am probably incorrect about fixing it with granting `sudo` though. It might vary from system to system

LongLiveCHIEF, 2020-08-17:
What do you get when you run `ls /dev | grep tty`

cab404, 2020-08-17:
> What do you get when you run `ls /dev | grep tty`
> […](#)

In container: 
```
tty
ttyUSB0
```
On host 
```
tty
tty0..63
ttyS0..3
ttyUSB0
```

cab404, 2020-08-17:
GID for ttyUSB0 is 27 in both.

cab404, 2020-08-17:
[image]
I guess GID 27 just turned out to be `sudo` inside a container, that's why granting sudo is not a universal solution. 
I won't be making a network accessible container with python2.7 inside privledged either)

LongLiveCHIEF, 2020-08-17:
Yeah, each OS has slightly different udev rules for serial, character, and USB devices.  Adding to the group inside the container does nothing really, it's all about the host.

LongLiveCHIEF, 2020-08-17:
I will be including a section on setting up your host in the new docs, and this is a good reminder to put in a section around groups and permissions of the user running octoprint and those of the device attached to the container.

closed 2020-08-17 by LongLiveCHIEF

LongLiveCHIEF, 2020-08-17:
Also, don't think that just because I closed this PR that your contributions aren't welcome. I just want to make sure we know what problems we're solving so we can consider what problems we might also be causing so that we can add those things to our testing (which is also coming soon). :smiley:

There is a `dev-docker` channel on the official [OctoPrint Discord](https://discord.octoprint.org) if you ever want to chat in real time, or consult on development.

cab404, 2020-08-17:
I understand, really)

LongLiveCHIEF, 2020-08-17:
You may want to also consider trying out the new `octoprint/octoprint:1.4.2-camera` image variant we just published yesterday.  It uses a different permissions model that we'll be bringing to the main image variants over the next few days.