OctoPrint/octoprint-docker #71: Settings are not persisted between restarts

donfrigo, 2020-09-03:
Hello,

I am using the octoprint-docker image with the default docker-compose file, which specifies a volume.

When starting octoprint with the command: 
`docker-compose up -d`
it starts up normally and I am greeted with the setup.

If I stop it using:
`docker-compose down`,
it stops gracefully. 

Upon restarting it again, I am greeted with the same setup that I have completed earlier and all my settings are lost. What am I doing wrong? Thanks!

atom-b, 2020-09-06:
I submitted pull request #72 to fix this. It's a very simple change to docker-compose.yml that you can make yourself. 

If your container is still running and you want to save your current config you have to copy `/home/octoprint/.octoprint` from the container onto your host. You can do this with `docker cp octoprint-docker_octoprint_1:/home/octoprint/.octoprint .`. If your container isn't named `octoprint-docker_octoprint_1` then replace that with whatever your container name is, you can find that with `docker ps -a`.

After stopping your current container you'll need to delete the existing octoprint volume. Run `docker volume rm octoprint-docker_octoprint` to do that. In case your volume isn't named that, run `docker volume ls` to find it; it should have `octoprint` in the name.

LongLiveCHIEF, 2020-09-07:
this is actually due to a bug in the current build that I'll be fixing.  For now, the indicated workaround will work.  I'm not going to accept the PR, because the final build image when the bug is fixed actually symlinks `/home/octoprint` to `/octoprint`.

commit 080e0845 (2020-09-08): add octoprint

fixes #73
fixes #71
PR #74 (2020-09-08): Add octoprint back to base image
closes #73
closes #71
closed 2020-12-06 by LongLiveCHIEF