OctoPrint/octoprint-docker #80: raspicam 2.1 parameters

remb0, 2020-09-11:
I have installed the docker container with compose and everything works except my raspberry cam.
I think it's my fault because I don't know the richt settings:
in compose:
/dev/ttyACM0:/dev/ttyACM0

in the container settings:
CAMERA_DEV:/dev/ttyACM0
CAMERA_DEV:MJPEG_STREAMER_INPUT -y -n -r 640x48

but can't get it working. what do I wrong?

LongLiveCHIEF, 2020-09-11:
You usually don't need to set `CAMERA_DEV` at all. The defaults will work for most people right out of the box.

remb0, 2020-09-11:
thank you very much for your fast answer. 
I changed the ttyacm0 from stack to /dev/ttyvideo0
and removed the camera-dev.

I had it working on my raspberry without docker. but I decided to try to update python to 3.8. after that octoprint wasn't working and wanted to try the docker system.

But now I stuck into a working octoprint without camL
`Init v4L2 failed !! exit fatal

 i: init_VideoIn failed

MJPG Streamer Version.: 2.0

 i: Using V4L2 device.: /dev/video0

 i: Desired Resolution: 640 x 480

 i: Frames Per Second.: -1

 i: Format............: YUYV

 i: JPEG Quality......: 80

 i: TV-Norm...........: DEFAULT

libv4l2: error setting pixformat: Invalid argument

Unable to set format: 1448695129 res: 640x480

Init v4L2 failed !! exit fatal

 i: init_VideoIn failed`

LongLiveCHIEF, 2020-09-11:
Did you look at the docker-compose.yml file? https://github.com/OctoPrint/octoprint-docker/blob/master/docker-compose.yml

closed 2020-09-11 by LongLiveCHIEF

remb0, 2020-09-12:
I have tried the docker compose indeed, and I have nothing changed to it. so that's default.
but I see in my loggings:   i: Using V4L2 device.: (null)

can you please tell me what I have to check change or look at? 
I know that it's not a error in your work but a failure of my knowledge.   Thanks for you patience and help.

LongLiveCHIEF, 2020-09-12:
From your original post it looked like you were mixing up configuration settings that probably don't need to be changed for your hardware.

The most likely cast is that you need to add `start_x=1` kernel boot parameter to your raspberry pi to get your camera to work. https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md. After doing that, reboot, and try again, making sure you have `/dev/video0:/dev/video0` as a `device` in your compose file.

If that doesn't work, can you join either the discord.ocotprint.org [#support-docker](https://discord.gg/YmrznUe) channel, or the [octoprint support forum](https://community.octoprint.org/)?  This is for bugs and development, and those other 2 places are the best way for me to help you in real time.