OctoPrint/octoprint-docker #90: Restart of Octoprint within the App

toofewacres, 2020-09-27:
What server restart command works to restart the octoprint instance that can be put into the Server control options?

ax42, 2020-09-28:
I think this is tricky, as you need to restart the container which is a command on the host.  FWIW I've spun up a portainer instance so I can restart containers via a browser https://www.portainer.io/installation/

toofewacres, 2020-09-28:
That's what I do for now also. I also run Home Assistant and am able to have "switch" which turns docker containers off and on. Works for now.
I used to use a previous docker image (can't remember the name at the the moment) but you could do "supervisorctl docker restart octoprint"  and it would restart the docker container from within Octoprint.

LongLiveCHIEF, 2020-09-28:
This image is actually based on the image you're referring to, but uses a supervisor designed for containers instead of full operating systems. 

The command _should_ be (i actually haven't tested it yet):

```
s6-svc -r /var/run/s6/octoprint
```

toofewacres, 2020-09-28:
Tried your command and it did not work.

toofewacres, 2020-09-29:
I did some digging around and came up with this command line that worked:

redirfd -w 2 /dev/null s6-svscanctl -t /var/run/s6/services

There may be a better way to do this, but this is what I found that works.

LongLiveCHIEF, 2020-09-29:
I'm going to reopen if you don't mind, I'd still like to add a restart command suggestion to the documentation.

LongLiveCHIEF, 2020-09-29:
Just leaving some thoughts here for posterity... @toofewacres to be clear I'm not asking you to respond (unless you want to :smile:)

As @ax42 mentioned It's designed to just have a container restarted by the host as the best way, especially if you're doing something that changes the container environment, as a restart is the only way to load updates to the container environment.

The command posted works to restart _all_ services (not just octoprint), but may not have the same effect that restarting octoprint would on a normal operating system (for example after plugin update or intallation), and I'm afraid that could cause problems for users who aren't aware of this.

Almost thinking we need a section in the docs for octoprint-docker about "known differences/issues using octoprint in a container".

That being said, it would be worth investigating ways to trigger a container reload from within the container... although that comes with it's own challenges for container permissions/docker-in-docker types of scenarios.

One possible way would be if the hosts daemon was configured for the api to be acccessible on a port (by default it binds to a fd), and the container could send an http request from within the container to the api to trigger restart of the container.

The other approach is to just create a [finish script for the octoprint service that causes the container to shut down](https://github.com/just-containers/s6-overlay#writing-an-optional-finish-script), then a command to kill the octoprint service would be the server restart command you would configure... and as long as you're using a `restart` option in docker/compose, the container should automatically spin back up.

That last one is the approach I'm going to start with and will go from there.

toofewacres, 2020-09-29:
If there is a cleaner way of restarting just the Octoprint service, I'm all for it. I searched around last night and I guess I found the brut force method! I'm not all that familiar with the entire docker environment, but love the ease of installation, etc. that it gives you.

ertraid, 2020-10-03:
As per @toofewacres above,  the nunofgs/docker-octoprint image has the "supervisorctl docker restart octoprint" option.
Not sure what the technical shortcomings might be, but from a dumb user perspective I can say it works like a charm..

LongLiveCHIEF, 2020-12-06:
Once #132 is accepted, the command to restart octoprint will be preconfigured and work automatically for the docker image. Just like usual, the user will be able to view or change the restart command from the settings menu.

closed 2020-12-06 by LongLiveCHIEF

Thermionix, 2020-12-09:
I needed to set the restart command to `redirfd -w 2 /dev/null s6-svscanctl -t /var/run/s6/services`
The default `s6-svc -r /var/run/s6/octoprint` did not work for me

LongLiveCHIEF, 2020-12-09:
@Thermionix have you tried it with the latest updates?  I remembered you said this before from the forum, but that was before I added octoprint to the service directory.

oerkel47, 2020-12-10:
Hi. I started a clean container with the recent release and there was no preconfigured text for the restart setting. I tried both settings that @Thermionix mentionend and can confirm only the first once worked.

Ocoprint output:

`2020-12-10 20:21:55,463 - octoprint.server.api.system - INFO - Performing command for core:restart: s6-svc -r /var/run/s6/octoprint
octoprint        | 2020-12-10 20:21:55,511 - octoprint.server.api.system - WARNING - Command for core:restart failed with return code 111:
octoprint        | ! STDOUT:
octoprint        | ! STDERR: s6-svc: fatal: unable to control /var/run/s6/octoprint: No such file or directory
`

LongLiveCHIEF, 2020-12-12:
The default is not`s6-svc -r /var/run/s6/octoprint`.  It is: `s6-svc -r /var/run/s6/services/octoprint`

oerkel47, 2020-12-12:
Uh, that works. Thank you. Fyi, like I said it wasn't preconfigured and I didn't find this command anywhere, maybe add it to readme :)

LongLiveCHIEF, 2020-12-12:
yeah, i will be adding it in there. I've noticed there are a few situations when the config doesn't get pre-loaded, and I'm still working through those as well.

commit 1d83247d (2020-12-06): preconfigure docker specific settings

closes #90