OctoPrint/octoprint-docker #93: Document ways to backup using docker

LongLiveCHIEF, 2020-09-29:
related #92 

Regardless of if we wind up implementing support of the bundled backup restore plugin, we should document a recommended/alternative method for backups of both the container and the volume using `docker` commands.

LongLiveCHIEF, 2020-09-29:
Just in case someone else gets to this before I do, here's some info that will help put together this information:

Docker has a a few good reference on this in their docs
- [Docker Docs | Backup, Restore, or Migrate Data Volumes](https://docs.docker.com/storage/volumes/#backup-restore-or-migrate-data-volumes)
- [Docker Docs | `export` reference](https://docs.docker.com/engine/reference/commandline/export/)

...and this article from geek diary,[How to Backup and Restore Docker Containers](https://www.thegeekdiary.com/how-to-backup-and-restore-docker-containers/) also has a great synopsis

atxbyea, 2021-02-26:
Perhaps also mention about backup up persistent volumes ? I can do that if wanted, with examples of rsync \ rclone \ rsnapshot etc.