OctoPrint/octoprint-docker #97: Q: How do I disable mjpg-streamer

freakadings, 2020-10-17:
## Question/Problem Description

I need to disable jmpg-steramer since i don't want to use a webacam and my log gets spamed with:

```
MJPG-streamer [11840]: TV-Norm...........: DEFAULT
 i: init_VideoIn failed
```

if It's possible via an environment variable it would be great.

### System Details

I'm running octoprint on a Synology Discstation 218+
```

Include the output of `docker version` (and `docker-compose` version if using compose):

Client:
 Version:           18.09.8
 API version:       1.39
 Go version:        go1.11
 Git commit:        bfed4f5
 Built:             Fri Mar 13 06:46:11 2020
 OS/Arch:           linux/amd64
 Experimental:      false

Server:
 Engine:
  Version:          18.09.8
  API version:      1.39 (minimum version 1.12)
  Go version:       go1.11
  Git commit:       3a371f3
  Built:            Fri Mar 13 06:44:35 2020
  OS/Arch:          linux/amd64
  Experimental:     false

```

## Describe your hardware

see above

pfeerick, 2020-10-17:
Try re-pulling the latest docker image for octoprint - Chief has just updated the images as the v2 image wasn't on dockerhub.

freakadings, 2020-10-18:
It seems to have worked. Thank you very much :)
(maybe you want to consider such an option for a future release, though)

closed 2020-10-18 by freakadings