OctoPrint/octoprint-docker #102: [BUG] 'latest' is pointing to pre-releases

LongLiveCHIEF, 2020-11-18:
The `latest` tag is only meant to point to full releases, but currently the workflows are considering pre-releases as `latest`.

LongLiveCHIEF, 2020-11-18:
It looks like this ran when I was testing changes to workflows, and was not triggered via dispatch.

jwaes, 2020-11-24:
Having the same issue ... 'watchtower' upgraded my image to :latest, then i got a message inside octoprint to 'upgrade' ... which was a downgrade to 1.4.2 and that fscked up my install.

LongLiveCHIEF, 2020-11-25:
>So the issue is that the jquery files are no longer there in the 'downgrade', as they moved to 1 js file ?

@jwaes this is most likely not a problem exclusive to docker, but I'll take a look.

commit 46f210e2 (2020-11-28): prevent release workflows from trigger on forks

fixes #102
commit 8b465a8c (2020-11-28): prevent release workflows from trigger on forks

fixes #102
closed 2020-11-28 by LongLiveCHIEF