OctoPrint/octoprint-docker #103: [BUG] Bug Description

Ilruz, 2020-11-23:
**Describe the bug**
Recommended docker compose file at https://github.com/OctoPrint/octoprint-docker/blob/master/docker-compose.yml won't work under the latest docker compose. Can also be easily fixed and improved to support permanent plugins, between images.

**To Reproduce**
Steps to reproduce the behavior:
1. Curl locally https://github.com/OctoPrint/octoprint-docker/blob/master/docker-compose.yml
2. execute docker-compose up --detach
2. Docker error will fire immediately, as 
 ports:
      - 80:80
is expecting a string.

**Expected behavior**
1. Docker container won't start, due to some 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
docker --version
Docker version 19.03.13, build 4484c46

docker-compose  --version
docker-compose version 1.27.4, build unknown

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

new suggested docker-compose.yml
version: '2.4'

services:
  octoprint:
    container_name: octoprint
    image: octoprint/octoprint
    restart: unless-stopped
    ports:
    - "80:80"
    devices:
    - /dev/ttyUSB0:/dev/ttyUSB0
    # - /dev/ttyACM0:/dev/ttyACM0
    # - /dev/video0:/dev/video0
    volumes:
     - octoprint:/octoprint

    # This volume will make the plugin permanent during restarts or image upgrade
     - octoprint-packages:/usr/local/lib/python3.8/sitepackages

    # uncomment the lines below to ensure camera streaming is enabled when
    # you add a video device
    #environment:
    #  - ENABLE_MJPG_STREAMER=true

  ####
  # uncomment if you wish to edit the configuration files of octoprint
  # refer to docs on configuration editing for more information
  ####

#config-editor:
  #  image: linuxserver/code-server
  #  ports:
  #    - 8443:8443
  #  depends_on:
  #    - octoprint
  #  restart: unless-stopped
  #  environment:
  #    - PUID=0
  #    - GUID=0
  #    - TZ=America/Chicago
  #  volumes:
  #    - octoprint:/config

#below directives will create a volume only if does not exist already
volumes:
  octoprint:
    external: false
  octoprint-packages:
    external: false

LongLiveCHIEF, 2020-11-25:
in regards to the 'expected string or buffer' issue for ports, this is indicative of a distro issue, see: https://github.com/docker/compose/issues/4972

in regards to the making plugins persistent, that is a super awesome suggestion, and I'll probably look at working it into the work I'm doing on the https://github.com/OctoPrint/octoprint-docker/tree/run-s6-as-octoprint-user branch.

LongLiveCHIEF, 2020-11-25:
we do have `PYTHONUSERBASE` set to `/octoprint/plugins` in the Dockerfile, so if i'm understanding python correctly, that _should_ make the site packages directory relative to that path.  It's possible though that since we're using s6, that this env is not loaded.

This may actually be a _bug_, but I'll have to do more digging.

LongLiveCHIEF, 2020-12-09:
closed via #132

closed 2020-12-09 by LongLiveCHIEF