OctoPrint/octoprint-docker #133: investigate supporting release channels for corresponding tags

NotExpectedYet, 2020-12-07:
**Is your feature request related to a problem? Please describe.**
I currently use bare installs for my testing farm and I believe this would be much simpler and convenient to move this over to docker. Currently the docker version is lacking some settings that I would personally like, hopefully they will benefit others. 

**Describe the solution you'd like**
I would like the ability firstly to set a username and password that is automatically setup on boot, this could be further expanded with all the settings required for the initial wizard to be skipped. This could be provided with a config.yaml file, or enviroment variable with docker that the initial script generates on boot. 

**Describe alternatives you've considered**
I already currently do something similar with my current setup but would be impossible for this docker currently without firstly generating my own version of your docker image and using that, then my own tags ontop. I have no qualms doing that personally but I think it would be beneficial for other users of OctoPrint to have them here. I know some use dockers with my OctoFarm setup and due to the recent locking down of user accounts may make their personal setup times much faster whilst keeping OctoPrint's initial secure nature. 

Let me know if anything doesn't make sense I'm only two coffee's deep.

LongLiveCHIEF, 2020-12-08:
related to #20. I'm working on this in the https://github.com/OctoPrint/octoprint-docker/tree/run-s6-as-octoprint-user branch, but that branch may get renamed/reworked a bit to support a username that is defaulted to octoprint, but is provided/overridable by the user.

By default the image will drop privs for this user except for where it's needed for octoprint.

For now, s6 would still be run as root, but the octoprint service would be run as a non-privileged user.

LongLiveCHIEF, 2020-12-08:
whoops, commented on the wrong issue, ignore that previous one.

I've been thinking about this also, just because it's needed to really do real e2e testing. Def on the roadmap, but probably won't be able to start for another few weeks.

S6 gives us options here to inject the container environment into the service configuration, which I've already done with MJPG_STREAMER, so I'll be utilizing a similar approach for octoprint where possible. 

The first piece of this was implemented already, by providing a configuration.yaml file, and you can also use the provided config editor service to pre-configure if you know your values.

The one that I'm trying to really look into since it will be the biggest challenge is providing those credentials.  My guess is that there is a command line option, but I haven't had a chance to look yet.

NotExpectedYet, 2020-12-11:
> whoops, commented on the wrong issue, ignore that previous one.
>
> I've been thinking about this also, just because it's needed to really do real e2e testing. Def on the roadmap, but probably won't be [...]

Epic cheif, thanks for the response. 

Could you explain the last bit for me a bit more please? I gather you mean providing the initial login variables to OctoPrint just wanted to clarify aha. If so, an inject into the config.yaml / user.yaml works from past experience.

LongLiveCHIEF, 2020-12-11:
I'm talking about application user and credential creation control via container configuration... assuming we can do it in a reasonably secure way.

This is distinct from the user/permissions needed to run octoprint itself.

LongLiveCHIEF, 2020-12-19:
I'm thinking of creating a `OCTOPRINT_STARTUP_FLAGS` env var that passes input to `octoprint serve`.  
For anything else i could add documentation around configuration strategies:
- build your own Dockerfile from the base image, passing your own `config.yaml` during build
- docker exec into the container and use the [octoprint cli](https://docs.octoprint.org/en/master/configuration/cli.html) to change configuration
- enhance docs on utilizing the config-editor service provided in the docker-compose.yml prior to starting octoprint service/reloading the octoprint service after updating config.yml

LongLiveCHIEF, 2020-12-20:
o i've done some initial work on the configuration issue you opened.  I'm kinda torn on providing env vars for things that go in config.yaml, since the main octoprint doesn't support them.  I feel like if that's a feature to add, then it should actually be requested upstream and then we can support it in docker.

I'm not really sure there's value in that though, since you can set these values with the command line.

It's docker best practice to build on images of others to add your specializations, and this fits the pattern.  You can either mount in a config.yaml during build of your own image, or mount in a config.yaml during container startup.

You can also utilize docker secrets if you do the "use octoprint/octoprint as a base" technique, and I actually added some demonstration of that in the latest pull request #153 

I'll continue to add docs for this, but I think for now this issue is out of scope of the docker image.

closed 2020-12-20 by LongLiveCHIEF