OctoPrint/octoprint-docker #137: [BUG] Bug Description

ertraid, 2020-12-11:
**Describe the bug**
After upgrade to 1.5.1 I cannot get to the UI… I can still connect with OctoPod, see camera, home, etc… but if logging to the UI it gets stuck on "Connecting to Octoprint Server"… tried adding /recovery/ and I get "Server Error".

**To Reproduce**
Steps to reproduce the behavior:
1. Start form working 1.4 installation on RPI4/docker
2. Upgrade to 1.5.1 through normal UI process
3. See error

**Expected behavior**
Continued work under new version (1.5.1)

*Additional context**
Quoting Charlie_Powell [post:3, [topic:27202;]](https://community.octoprint.org/t/issue-upgrading-1-4-to-1-5-1/27202)
"This is a reoccurring issue with docker, it seems to miss updating some of the JS libraries on OP update. Strangest thing is that not only is it just the JS files, it's always jQuery UI that fails too. You can see these 'file not found' errors in your log."

ertraid, 2020-12-12:
Not sure if related or not, but just noticed at some point today my container (running octoprint/octoprint:latest) has been pushed to 1.5.2 and the new folder structure with no apparent issues...

LongLiveCHIEF, 2020-12-14:
I don't know what's causing this, but I just got this same issue on an instance of OctoPrint that doesn't have docker anywhere in sight, and have also seen several users in the octopi discord channel reporting the same thing. So closing here, and recommending this be opened in octoprint/octoprint.

closed 2020-12-14 by LongLiveCHIEF