OctoPrint/octoprint-docker #138: [BUG] Data directory changed to /octoprint/octoprint ?

morganchristiansson, 2020-12-11:
**Describe the bug**
After `docker-compose pull && docker-compose up -d` all my data was gone. Looking inside the container it's created new folder structure under `/octoprint/octoprint` whereas previously they were in `/octoprint`

**To Reproduce**
```
docker-compose exec octoprint1 bash -l
ls -l /octoprint/octoprint
```

**Expected behavior**
Files used to live directly under /octoprint

** Workaround **
Moving everything to /octoprint/octoprint and plugins stopped working. I guess they should still be in /octoprint/plugins ?

To make things simple I just created symlink using `ln -s . octoprint` inside /octoprint (outside or inside docker)

**Desktop (please complete the following information):**
 - OS: Raspbian
 - Browser n/a
 - Version 1.5.2 / 1.5.x

cp2004, 2020-12-12:
I think this is detailed in the release notes [V3.0.0](https://github.com/OctoPrint/octoprint-docker/releases/tag/3.0.0)

I'll leave someone else to figure out if it is an actual bug, but just wanted to make sure you had read them to see if this explains the behaviour.

LongLiveCHIEF, 2020-12-12:
Yeah, I was afraid of this, which is why the latest release was a breaking change. I readily admit I should have done a better job of identifying the impact and mitigation steps.  I'll get started on that now.

LongLiveCHIEF, 2020-12-12:
Just FYI moving everything from `/octoprint/octoprint` was the right instinct, and there was just a missing second step, which was moving the `/octoprint/octoprint/plugins/python3.8` directory to `/octoprint/plugins`.  that directory contains the actual plugin binaries

LongLiveCHIEF, 2020-12-12:
> An upgrade note / breaking changes note in README would've been nice

There are breaking change notes in both the CHANGELOG and in releases.  Even if I put the notes in README, you wouldn't see them before you updated. Whereas with the breaking changes being in the changelog and in releases, gitlab has built-in notification functionality for you by clicking Watch > custom > releases.

In addition to that, I have pinned in both docker channels in discord when there are breaking changes.

Users need to take some responsibility to check for changes before pulling new images, and I'm doing everything I can to push notifications in multiple places and ways.

LongLiveCHIEF, 2020-12-13:
@morganchristiansson I've made a change to the README in PR #142 that contains the following language. Can you read it over and let me know if you think that would have worked for you had it been there beforehand?

> The `octoprint/octoprint` image uses semantic versioning, but the tags for `octoprint/octoprint` follow the
version of octoprint contained in the image. As a result we recommend you always check the [CHANGELOG](CHANGELOG.md)
or [Releases](https://github.com/OctoPrint/octoprint-docker/releases) before pulling an image, _even if you are pulling the same tag_.

>You can subscribe to be notified of releases as well, by selecting the Watch button in the upper right corner, 
choosing "Custom", and checking "Releases".

>In addition, we know that OctoPrint is not the best suited type of application for containerization, but we're
working hard to make it as compatible as possible. Please check out our [Roadmap](https://github.com/OctoPrint/octoprint-docker/projects/4),
or join the discussion in the `#dev-docker` or `#support-docker` channels on the official OctoPrint Discord
[discord.octoprint.org](https://discord.octoprint.org).

PR #142 (2020-12-13): V3 image migration
- closes #141 
- closes #138 

Instead of creating a migration strategy guide in the release docs, I went the route of detecting when a v3 container was being started with a v2 file structure, and then auto-migrate the file structures for them during container initialization.

This would potentially be a breaking release, in the event that two octoprint containers are using the same volume and the second container is added while the first (which is a v2 container) is running.

However, I've decided against classifying this as a breaking change (and thereby requiring a v4 rel [... cut]
closed 2020-12-13 by LongLiveCHIEF