OctoPrint/octoprint-docker #144: CONTRIBUTING.md file missing from project (README.md has link to it)

justindelliott, 2020-12-13:
The [README.md](https://github.com/OctoPrint/octoprint-docker/blob/master/README.md) mentions referring to the CONTRIBUTING.md file for details on how to build the image yourself:

**If you would like to build the docker image yourself, please read [CONTRIBUTING.md](https://github.com/OctoPrint/octoprint-docker/blob/master/CONTRIBUTING.md) for details on how to do so.**

Unfortunately the link for CONTRIBUTING.md is broken:

https://github.com/OctoPrint/octoprint-docker/blob/master/CONTRIBUTING.md

Can you add the file back to this repo or update the link to the new file / path? (I looked around the files in the master branch before submitting this but couldn't find it.)

Thanks!

LongLiveCHIEF, 2020-12-14:
Good catch, I just removed this in a recent pull because the automation it had in it was no longer used so it served no purpose. I need to get new CONTRIBUTING docs out there though.

LongLiveCHIEF, 2020-12-14:
For reference, here's the _old_ CONTRIBUTING https://github.com/OctoPrint/octoprint-docker/blob/974a677eb04511377f140625eff9ee70813e87de/CONTRIBUTING.md

The new contributing should cover:
- [  ] How to test multi-arch builds
- [  ] How to join #dev-docker channel in discord
- [  ] How the [Roadmap](https://github.com/OctoPrint/octoprint-docker/projects/4) works
- [  ] If PR's fix a problem/bug, there should be a corresponding issue created for it first
- [  ] PR's for enhancements/typos/QoL stuff are ok without having corresponding issues
- [  ] How to test/setup workflows/actions

It's early morning Monday as I type this so I know I'm probably missing something, and any input from contributors as to what should go on this new CONTRIBUTING as well would be welcomed!

justindelliott, 2020-12-15:
Got it, thanks!

PR #151 (2020-12-19): Create local build test utils
- his adds some utilities to make building, testing, and configuring your build system easier.
- Adds documentation for using these utilities in CONTRIBUTING
- Adds documentation for PR guidelines in CONTRIBUTING (closes #144)
closed 2020-12-19 by LongLiveCHIEF