OctoPrint/octoprint-docker #145: Add package cpulimit to build

M1XZG, 2020-12-14:
At least one plugin, [octoprint-telegram](https://plugins.octoprint.org/plugins/telegram/), requires the use of the package `cpulimit` in order to produce it's gifs for telegram notifications.

Could you please add this package to the base image via the dockerfile? I'm happy to manage my own dockerfile but would be nice if it was part of the base. 

Error from [octoprint-telegram](https://plugins.octoprint.org/plugins/telegram/) is:

```
😵 Problem creating gif, please check log file, and make sure you have installed cpulimit with following command : `sudo apt-get install cpulimit`
```

Thanks

LongLiveCHIEF, 2020-12-14:
I have docs coming as we speak on how to build your own image from the base octoprint image.  The general answer to "plugin needs some OS package" will likely be to manage this with your own docker build, however I'll consider each request based on these merits:

- is the plugin so common that most users will need this package
- is the package available in all distros & architectures that the `octoprint/octoprint` image is distributed on

@M1XZG I'll do the research on this one and let you know what I find. My biggest concern with this one is making sure it is available on `arm/v7` and `arm64`.

M1XZG, 2020-12-14:
Thanks, I was just about to edit this and add another package to the base list too... `wget` would be useful as a number of plugins can use this for API calls (ifttt for example)

LongLiveCHIEF, 2020-12-14:
yeah, `wget` is something I don't have any concerns about adding :smiley:

LongLiveCHIEF, 2020-12-14:
Maybe another option would be to add an ENV for `ADDITONAL_PACKAGES`, which would be a comma-separated list of values the user can pass in, and these packages would be ensured during container initialization. :thinking:

LongLiveCHIEF, 2020-12-14:
we can continue _that_ discussion in discord for now though :smirk: there are pros/cons to that approach.

LongLiveCHIEF, 2020-12-20:
So the decision on this one is to leave it out. Docker containers are designed so the processes running in them don't restrict themselves, and the cpu limits of the container are managed by the docker daemon.

Adding a package into the main build that could conflict with services that might manage container cpu limits could cause problems. (Home Assistant, OctoFarm, Portainer, etc...)

closed 2020-12-20 by LongLiveCHIEF