OctoPrint/octoprint-docker #147: Rescan .gcode file uploads after upgrading OctoPrint container to a newer version?

justindelliott, 2020-12-16:
Is there a way to force OctoPrint to rescan the `./octoprint/uploads` directory after upgrading the container from an earlier version of OctoPrint so all of the previous .gcode uploads are still available within OctoPrint? Otherwise I have to upload all of the .gcode files again.

After I upgraded (deleted my local container, pulled the updated container) all of my previous .gcode uploads no longer appear in the OctoPrint Files section - but they still exist on the host's local file system.

It's good that the local `uploads` directory is not touched, but I wonder if that could be a problem when OctoPrint isn't aware of the pre-existing files and tries to replace them when a new .gcode is uploaded?

Here's my docker-compose file, in case I'm missing anything or doing something stupid:

```shell
version: '3.7'

services:
  octoprint:
    image: octoprint/octoprint
    container_name: octoprint-mini
    ports:
      - 80:80
    environment:
      - CAMERA_DEV=/dev/video1
      - MJPEG_STREAMER_INPUT="-y -n -r 1280x720"
      - ENABLE_MJPG_STREAMER=true
    devices:
      - /dev/ttyACM_prusa_mini:/dev/ttyACM_prusa_mini
      - /dev/video1:/dev/video1
    volumes:
      - ./docker-volumes/octoprint:/octoprint
    restart: unless-stopped
volumes:
  octoprint:
```

Note that I've set up udev rules for my host running Centos 7 to name the printer.

Thanks for doing this project!

LongLiveCHIEF, 2020-12-16:
Probably. I'll add it to the roadmap, thanks!

justindelliott, 2020-12-16:
Cool! Thanks.

LongLiveCHIEF, 2020-12-19:
I think this is going to be as simple as changing the upload directory pre-configuration.

LongLiveCHIEF, 2020-12-19:
@justindelliott do you know what image version you're running?  I think the changes I recently made with 3.0.0 image version actually fixed this. If you upgraded from a v2 image to a v3 image, you may just need to restart the container with [automigration enabled](https://github.com/OctoPrint/octoprint-docker/blob/master/README.md#container-environment-based-configs) to restore gcode and timelapses.

Can you give that a shot and let me know?

justindelliott, 2020-12-19:
That worked! Would have helped if I had read all of the documentation! Sorry I wasted your time here.

For the benefit of others who might run into this same thing...

I added "AUTOMIGRATE=true" to my docker-compose.yml and all of the previous .gcode files were available after upgrading the container of OctoPrint from v1.4.2 to v1.5.2.

Thank you!

LongLiveCHIEF, 2020-12-19:
No problem! To be fair, even I didn't think about it, and I both broke the feature, and wrote the automigration feature. lol.

closed 2020-12-19 by LongLiveCHIEF