OctoPrint/octoprint-docker #159: Web interface not loading with OctoPrint in Docker on Raspberry PI4

TheNetAdmin, 2020-12-26:
**Describe the bug**

The web interface does not load with docker running on RPI4, similar to https://github.com/OctoPrint/OctoPrint/issues/3502

This issue is redirected from OctoPrint main repo https://github.com/OctoPrint/OctoPrint/issues/3907

**To Reproduce**

1. Install `docker-compose` on Raspberry PI 4
2. Use the `docker-compose.yml` provided by https://github.com/OctoPrint/octoprint-docker/blob/master/docker-compose.yml
3. Start the docker with `docker-compose up -d`, which uses the `latest` docker tag, referring to OctoPrint 1.5.2
4. Open the web interface on another desktop (under the same local network), the Raspberry PI is not connected to any printer yet

**Expected behavior**

The web interface should load.

**Screenshots**

The web interface report an error "Loading failed", similar to issue https://github.com/OctoPrint/OctoPrint/issues/3502

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 87.0.4280.88

**Additional context**

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5742916/octoprint.log)

[JavaScript Console in Chrome](https://github.com/OctoPrint/OctoPrint/files/5742918/192.168.50.5-1608976927375.log)

LongLiveCHIEF, 2021-01-01:
You say Raspberry Pi4, but then your OS you say Windows 10.  Is Windows 10 the machine you're accessing it from and the container is actually running on an RPi4?

@jneilliii was there something in particular that made you think this was a docker specific issue, or anything you can add that will help me with where to debug?

The good thing about containers is that they're static. Meaning I know _exactly_ what happens if you ran the docker-compose.yml from the example. Which means for this to happen, it has to be one of 3 things:
- user changed something
- application changed something

@TheNetAdmin can you help me answer a few questions?

1. are there any errors in the docker-compose logs?
2. was this a fresh install?
  - if no, When was the last time you ran `docker pull octoprint/octoprint`?

jneilliii, 2021-01-01:
I suppose it could be plugin related, but because they're using the docker-compose figured support should start here rather than core octoprint. I suspect the command to run in the container for safe mode would be `octoprint serve --safe --iknowhatimdoing` to troubleshoot possibility of plugin related issues.

LongLiveCHIEF, 2021-01-01:
> I suspect the command to run in the container for safe mode would be octoprint serve --safe --iknowhatimdoing to troubleshoot possibility of plugin related issues.

I'm going to have to look into how to start the container in safe mode. This won't work because... well... docker. I'll probably have to add a `SAFE_MODE` env var and update the service script to look for that env var and pass it to the service initialization script. This also means that the container would need to be restarted to reload the container environment, which means you couldn't restart octoprint from within octoprint.

> but because they're using the docker-compose figured support should start here

Docker images have static state, so in most cases (unless they're doing something non-standard) if they're using docker in means you can _eliminate_ docker/operating system/network. That's one of the advantages of docker, allowing you to eliminate a whole stable possible side effect originators, and isolate problems unique to the application.

TheNetAdmin, 2021-01-03:
> You say Raspberry Pi4, but then your OS you say Windows 10. Is Windows 10 the machine you're accessing it from and the container is actually running on an RPi4?

Yes, the `docker-compose` runs on RPi4 with the latest, freshly installed RPI OS. And I'm using Windows 10 Chrome to access it. Android Chrome gives the same results.

> 1. are there any errors in the docker-compose logs?

Here is my logs with the command `docker-compose logs` after running `docker-compose up -d` and using Chrome to access the web interface. `192.168.50.5` is the RPI4 IP, and `192.168.50.239` is Windows 10 IP.

```
octoprint_1  | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
octoprint_1  | [s6-init] ensuring user provided files have correct perms...exited 0.
octoprint_1  | [fix-attrs.d] applying ownership & permissions fixes...
octoprint_1  | [fix-attrs.d] done.
octoprint_1  | [cont-init.d] executing container initialization scripts...
octoprint_1  | [cont-init.d] 01-mjpg-streamer-config: executing...
octoprint_1  | [cont-init.d] 01-mjpg-streamer-config: exited 0.
octoprint_1  | [cont-init.d] v3-dir-structure-check: executing...
octoprint_1  | [cont-init.d] v3-dir-structure-check: exited 0.
octoprint_1  | [cont-init.d] done.
octoprint_1  | [services.d] starting services
octoprint_1  | [services.d] done.
octoprint_1  | 2021-01-03 04:22:10,976 - octoprint.startup - INFO - ******************************************************************************
octoprint_1  | 2021-01-03 04:22:10,977 - octoprint.startup - INFO - Starting OctoPrint 1.5.2
octoprint_1  | 2021-01-03 04:22:10,978 - octoprint.startup - INFO - ******************************************************************************
octoprint_1  | 2021-01-03 04:22:11,226 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
octoprint_1  | 2021-01-03 04:22:11,226 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
octoprint_1  | 2021-01-03 04:22:11,226 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
octoprint_1  | 2021-01-03 04:22:11,260 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python3.8/site-packages/octoprint/plugins, /octoprint/octoprint/plugins and installed plugin packages...
octoprint_1  | 2021-01-03 04:22:12,118 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
octoprint_1  | 2021-01-03 04:22:12,264 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 16 mixin implementations, 32 hook handlers
octoprint_1  | 2021-01-03 04:22:12,324 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
octoprint_1  | 2021-01-03 04:22:12,331 - octoprint.server - INFO - Intermediary server started
octoprint_1  | 2021-01-03 04:22:12,349 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /octoprint/octoprint/uploads...
octoprint_1  | 2021-01-03 04:22:12,350 - octoprint.filemanager.storage - INFO - ... file metadata for /octoprint/octoprint/uploads initialized successfully.
octoprint_1  | 2021-01-03 04:22:12,352 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
octoprint_1  | 2021-01-03 04:22:12,353 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
octoprint_1  | 2021-01-03 04:22:12,353 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
octoprint_1  | 2021-01-03 04:22:12,354 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
octoprint_1  | 2021-01-03 04:22:12,354 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_read'), Need(method='role', value='plugin_announcements_manage')")
octoprint_1  | 2021-01-03 04:22:12,354 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
octoprint_1  | 2021-01-03 04:22:12,355 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
octoprint_1  | 2021-01-03 04:22:12,355 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
octoprint_1  | 2021-01-03 04:22:12,355 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
octoprint_1  | 2021-01-03 04:22:12,356 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
[... cut ...]
```
docker-compose version 1.21.0, build unknown
docker-py version: 3.4.1
CPython version: 3.7.3
OpenSSL version: OpenSSL 1.1.1d  10 Sep 2019
```

> 2. was this a fresh install?
> 
> * if no, When was the last time you ran `docker pull octoprint/octoprint`?

Yes it's a fresh install. I installed the RPI OS (previously called Raspbian) latest version. Then installed `k3s` kubernet (but I stop and disable it through `systemctl`). Then install `docker-compose` with `octoprint` image latest version.

Acelogic, 2021-01-03:
@TheNetAdmin Wow-what great timing, I'm experiencing the same issue on my Raspberry PI 4 also. I'm looking into what could possibly be the issue.

Acelogic, 2021-01-03:
So I have an update, I made the ports in the docker compose file to use 5000:5000 and it seems like the docker container is working now. However, whenever I set a custom port pair it doesn't work anymore.

TheNetAdmin, 2021-01-03:
> So I have an update, I made the ports in the docker compose file to use 5000:5000 and it seems like the docker container is working now. However, whenever I set a custom port pair it doesn't work anymore.

Oh YES, `5000:5000` works for me. `80:5000` also works.

LongLiveCHIEF, 2021-01-03:
If port 5000 is working, that means you're either using a cached version of the octoprint image, or the tag for `latest` for your architecture is not getting updated. What happens when you explicitly use `octoprint/octoprint:1.5.2` instead of `octoprint/octoprint`?

LongLiveCHIEF, 2021-01-03:
And also, if both of you could run a `docker image inspect octoprint/octoprint:latest` for me and paste results I'd appreciate it.

LongLiveCHIEF, 2021-01-03:
I believe #162 should fix this.

LongLiveCHIEF, 2021-01-03:
I'm closing this with #162.  If after pulling the new image this still happens, then re-open.

Thanks! :heart:

closed 2021-01-03 by LongLiveCHIEF

TheNetAdmin, 2021-01-09:
> And also, if both of you could run a `docker image inspect octoprint/octoprint:latest` for me and paste results I'd appreciate it.

Sorry for the late reply, here's the output

```
$ docker image inspect octoprint/octoprint:latest
[
    {
        "Id": "sha256:83831e1b5a6cde49baa7f894091e45d8115a2358dae7244db6c89fd453a67183",
        "RepoTags": [
            "octoprint/octoprint:1.5",
            "octoprint/octoprint:latest"
        ],
        "RepoDigests": [
            "octoprint/octoprint@sha256:ea5f88138a014893c87f40df8476e4f0e9c55b46e3538cdf92f3e8cfcfc716f9"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2020-12-20T15:18:47.808777194Z",
        "Container": "",
        "ContainerConfig": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": null,
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": null,
            "Labels": null
        },
        "DockerVersion": "",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "ExposedPorts": {
                "80/tcp": {}
            },
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/octoprint/plugins/bin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "LANG=C.UTF-8",
                "GPG_KEY=E3FF2839C048B25C084DEBE9B26995E310250568",
                "PYTHON_VERSION=3.8.6",
                "PYTHON_PIP_VERSION=20.3.3",
                "PYTHON_GET_PIP_URL=https://github.com/pypa/get-pip/raw/5f38681f7f5872e4032860b54e9cc11cf0374932/get-pip.py",
                "PYTHON_GET_PIP_SHA256=6a0b13826862f33c13b614a921d36253bfa1ae779c5fbf569876f3585057e9d2",
                "octoprint_ref=1.5.2",
                "CAMERA_DEV=/dev/video0",
                "MJPG_STREAMER_INPUT=-y -n -r 640x480",
                "PIP_USER=true",
                "PYTHONUSERBASE=/octoprint/plugins"
            ],
            "Cmd": null,
            "Image": "",
[... 35 lines cut ...]
                "sha256:b2922aff2910c1ed4b92a0710eb0d901d5d8c424225c4197f03ca4b5081cb460",
                "sha256:d9bdc92c409970aa9de880fd8533c5a88688eb5dc417f80c40de3e413b83c1ab",
                "sha256:2eac823d0b9da52e3d1d67a3f7a25d4f03e3b672975626477cb93ddd1d4107a9",
                "sha256:aeec1fb19fcef0e4bb817112556d7d2806bfe3512aaccb4a4cb543d79e93a3f5",
                "sha256:ee41b08858b105c95ee6fe59d35d0fcbd04dcaf7f11b4f0584b2a15bc8675cfb",
                "sha256:05e6afcb78d63205ec7bdaac253605549935076a0831e79b63a34234bbb5ca89",
                "sha256:16d4a35d943df93726a3bdacc5f5852a099b837ae6426d395ccdee3cd48c1866",
                "sha256:807a6253b41e3b72e2274521f5393a4f03d0dce45d814e67b5ebd3c2cf136fe4",
                "sha256:e3f556a9496f7580b970b0cbe1bc11372fe00c210d73fb465d0c957a7455f48e",
                "sha256:3e582b747bbbd3d17d2612789b610095208caf973989d37297b018b449f76e6f",
                "sha256:efe50368a2a7da8bd664b265ef9288d707b17c630edf74ff12232ddfff3fcc19",
                "sha256:7b23a8a98701c72f7e12c09360abd86b801c9fab1395f5f6d1d8014e54a26443",
                "sha256:63df2e01797fd8791623ecebc411258c261f401905f6516b9a86b1260d0089df",
                "sha256:930ce9ef57353915dd15d092b0c67b67ca17b0221b428aa1367015c7188b4dd7",
                "sha256:16d4b086ca31351337819391459d829d5893f59e4c04e17d0338fee08feeb03c"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
```

It seems I'm using `1.5` version as `latest`, if I'm not mistaken.

TheNetAdmin, 2021-01-09:
> If port 5000 is working, that means you're either using a cached version of the octoprint image, or the tag for `latest` for your architecture is not getting updated. What happens when you explicitly use `octoprint/octoprint:1.5.2` instead of `octoprint/octoprint`?

This problem persists with using `1.5.2` explicitly:

```yaml
version: '2.4'

services:
  octoprint:
    image: octoprint/octoprint:1.5.2
    restart: unless-stopped
    ports:
      - 80:80
    devices:
      - /dev/ttyUSB0:/dev/ttyACM0
    #  - /dev/video0:/dev/video0
    volumes:
     - octoprint:/octoprint
    # uncomment the lines below to ensure camera streaming is enabled when
    # you add a video device
    #environment:
    #  - ENABLE_MJPG_STREAMER=true

  ####
  # uncomment if you wish to edit the configuration files of octoprint
  # refer to docs on configuration editing for more information
  ####

  #config-editor:
  #  image: linuxserver/code-server
  #  ports:
  #    - 8443:8443
  #  depends_on:
  #    - octoprint
  #  restart: unless-stopped
  #  environment:
  #    - PUID=0
  #    - GUID=0
  #    - TZ=America/Chicago
  #  volumes:
  #    - octoprint:/octoprint

volumes:
  octoprint:
```

```
$ docker-compose up -d
Creating network "docker_default" with the default driver
Pulling octoprint (octoprint/octoprint:1.5.2)...
"docker-compose.yml" 39L, 855C
1.5.2: Pulling from octoprint/octoprint
c06905228d4f: Already exists
576e27635e2a: Already exists
31b618766132: Pull complete
ab17eeea6058: Pull complete
aa21e0c66711: Pull complete
e086ae249d6e: Pull complete
c625877dcc7e: Pull complete
264703e1c6cc: Pull complete
acf043c55166: Pull complete
b645e52cbc8c: Pull complete
cf39aa59da55: Pull complete
dcd87ad589fc: Pull complete
b56d00153465: Pull complete
617911dc3fbe: Pull complete
a73b3960836b: Pull complete
Digest: sha256:51c7647e5570d0a28a3b17d7dc055509931ada707f4ad87532a31e286bb45ffe
Status: Downloaded newer image for octoprint/octoprint:1.5.2
```

TheNetAdmin, 2021-01-09:
> I'm closing this with #162. If after pulling the new image this still happens, then re-open.
> 
> Thanks! ❤️

Thank you for the update, I will try the new image once it's released. (Or is it already released, and I can pull it?)

LongLiveCHIEF, 2021-01-09:
It's published.

averagenative, 2023-03-25:
This is still an issue