OctoPrint/octoprint-docker #160: [BUG] port is not a valid integer

monodop, 2020-12-26:
**Describe the bug**
Container shows as running, but the logs show the following error
```
Usage: octoprint [OPTIONS] COMMAND [ARGS]...
Try 'octoprint --help' for help.
Error: Invalid value for '--port': tcp://10.43.75.180:80 is not a valid integer
```

**To Reproduce**
I have configured this workload in rancher:
[image]

**Environment**
Ubuntu/Rancher 2 single-node cluster

**Additional information**
This error doesn't seem to happen if I don't try to bind the ports - in this case I get a lot more logs that seem to indicate that the application is running properly. However, it's not very useful if I don't bind the ports to the host.

bjw-s, 2020-12-29:
I'm seeing the same thing when I'm not setting the env var `OCTOPRINT_PORT: "5000"` explicitly.

monodop, 2020-12-29:
> I'm seeing the same thing when I'm not setting the env var `OCTOPRINT_PORT: "5000"` explicitly.

Hmm, I didn't see any mention of that environment variable in the docs, is that something I'm supposed to be configuring?

LongLiveCHIEF, 2021-01-01:
Are you overriding the default command or entrypoint, or mounting anything over any directories in `/etc`?

monodop, 2021-01-01:
> Are you overriding the default command or entrypoint, or mounting anything over any directories in `/etc`?

Nope, I believe everything is default except for what I showed in the screenshot

LongLiveCHIEF, 2021-01-01:
I don't have rancher, and have no way to test this. if you run it w/out rancher, w/out kubernetes, just using docker this doesn't happen. The signs point to somehow your container environment is being manipulated with your setup, because that's not an env var the docker container explicitly handles.

Also note, this container is really not meant to be used in a distributed system, and is not tested or supported on any runtimes other than docker. I can't make any guarantees about functionality on other runtimes. This container uses a special supervisor called s6-overlay, and it may not work exactly the same with the docker shim that kubernetes uses for containerd. (pretty sure rancher sits on top of k8s?)

All this is to say, if you're doing anything other than running the container on a single machine using docker directly, then I can't promise anything, and won't be able to promise much in the way of support.

It's not your typical docker container, but that's really because octoprint at it's core isn't really designed to support a distributed or containerized operating model.

I'll be happy to continue to toss ideas around, but unless we can confirm this happens in the supported setup and environment, I'm going to close the issue.

bjw-s, 2021-01-01:
Your post triggered me and made me dig in to it a bit more since I'm also running Octoprint in Kubernetes.

By default all services within a namespace are exposed to pods within that namespace through environment variables. If you happen to call your service `octoprint` (as we both probably did), you would get an environment variable called `OCTOPRINT_PORT` set to (e.g.) `tcp://10.43.32.6:5000`. This is however an environment variable that clashes with what Octoprint expects, causing the error we were seeing.

So there are two (maybe three) solutions:
- Explicitly set the OCTOPRINT_PORT env var to 5000
- Set another name for the Octoprint service
- Disable service links on the pod spec ([more info](https://kubernetes.io/docs/concepts/services-networking/connect-applications-service/#accessing-the-service))

monodop, 2021-01-01:
> Your post triggered me and made me dig in to it a bit more since I'm also running Octoprint in Kubernetes.
> 
> By default all services within a namespace are exposed to pods within that namespace through environment variables. If you happen to call your service `octoprint` (as we both probably did), you would get an environment variable called `OCTOPRINT_PORT` set to (e.g.) `tcp://10.43.32.6:5000`. This is however an environment variable that clashes with what Octoprint expects, causing the error we were seeing.
> 
> So there are two (maybe three, if you count renaming the service) solutions:
> 
> * Set another name for the Octoprint service
> * Explicitly set the OCTOPRINT_PORT env var to 5000
> * Disable service links on the pod spec ([more info](https://kubernetes.io/docs/concepts/services-networking/connect-applications-service/#accessing-the-service))

Ah, I didn't realize it was adding extra env vars based on the service's name. That fact might come in handy later on as well. I'll give these ideas a shot and see if I can get everything working this weekend.

LongLiveCHIEF, 2021-01-01:
> Your post triggered me...

Sorry! lol, I didn't mean to trigger you, however in my experience we learn the most and get the best feature enhancements when someone _does_ get triggered, or sets out to prove something the maintainers didn't believe could be done, so I appreciate your sacrifice!

I should note, that when I close an issue, it doesn't mean it's not worth discussing, it just means I believe the fix is with a 3rd party or putting in a shim would set a bad precedent for shimming 3rd party software.

bjw-s, 2021-01-01:
No harm done, I didn't mean I was triggered in a bad way, it just got my butt in gear to do more debugging 😄

monodop, 2021-01-01:
Maybe we don't need a fix directly in the image, but how do you feel about documenting the `OCTOPRINT_PORT` environment variable for the image documentation? I didn't realize it existed until I made this issue. Also are there other environment variables available?

LongLiveCHIEF, 2021-01-01:
I think the best practice is to explicitly document the env vars that are unique to the docker container, and then adding a note that "any environment variables supported by OctoPrint, HAPROXY, and mjpg-streamer are passed to their respective services".  

This is basic container principles stuff, so I don't want to re-document environment variables for the apps in the container because there'd be no way to know when they change.

The one thing I've run into in the last week, (and we've [had some discussion around it on discord](https://discordapp.com/channels/704958479194128507/705047010641838211/789933659922759720)), is that the few environment variables that octoprint ingests aren't in the official documentation. 

TL;DR: best practice for containers is to only document things unique to the container, but we should definitely open an issue in octoprint/octoprint for adding environment variable documentation.

LongLiveCHIEF, 2021-01-01:
So got to thinking...., setting the `OCTOPRINT_PORT` env var to anything _other_ than `5000` will break the container, since the haproxy cfg explicitly listens to that port.

I think there's an actual fix we need here, to set the `--port 5000` explicitly in the service start script so that `OCTOPRINT_PORT` is ignored if it's set.

PR #162 (2021-01-03): add port to init script
fixes #160 

Sets `--port 5000` in the octoprint service run script so that inadvertently setting `OCTOPRINT_PORT` doesn't break the haproxy integration.
closed 2021-01-03 by LongLiveCHIEF