OctoPrint/octoprint-docker #163: Enabling webcam in an existing container does not work even after container restart

LongLiveCHIEF, 2021-01-07:
Since `ENABLE_MJPG_STREAMER=false` removes the mjpg-streamer service entirely, simply restarting the container with that var set to `true` won't work, since s6-svscan won't pick up the service run script and start it up.

There is a workaround, by simply starting a _new_ container from the base image and attaching the old container volumes, which will work in the meantime until this feature can be added.

pfeerick, 2021-01-10:
For someone using docker-compose, `docker-compose down && docker-compose up -d` would be sufficient, wouldn't it?

Since, AFAIK, docker-compose down also removes the container? i.e. 

```
Stopping octoprint-docker_octoprint_1 ... done
Removing octoprint-docker_octoprint_1 ... done
Removing network octoprint-docker_default
Creating network "octoprint-docker_default" with the default driver
Creating octoprint-docker_octoprint_1 ... done
```

LongLiveCHIEF, 2021-01-12:
@pfeerick yep it would.

torresmvl, 2021-01-15:
@pfeerick @LongLiveCHIEF would  `docker-compose.override.yml` do the trick?

LongLiveCHIEF, 2021-01-17:
no, there's nothing special about that.

jferguson6, 2021-03-02:
Thanks for mentioning this issue - was trying to figure out why the camera wouldn't work. Ran
`docker-compose down && docker-compose up -d` and I could finally see the camera.

closed 2021-11-25 by LongLiveCHIEF