OctoPrint/octoprint-docker #166: Pulling a new version breaks setup/configuration

rvalle, 2021-01-08:
**Describe the bug**
I love octoprint and would like to stay updated, but it is proving very difficult becuase updating the docker image typically implies losing my settings. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install octoprint docker-compose
2. let some time pass
3. docker pull a new version of octoprint image
4. octoprint launches the user configuration, you lost your configuration
5. pull the previous image version (cancel the update)
6. octorpint launches user configuration, looks like the configuration has been lost permanently.

**Expected behavior**
After pulling a new docker version, ideally, everything should work and the old setup be functional. I understand this is an experimental feature (configuration migration), however, when not possible the following should be:
- new octoprint version detects and incompatible confirmation and stop without causing damage
- semantic versioning should be used, for example: 1.4.x configurations are always compatible, but updates to 1.5.x need migration (experimental) etc.

**Rationale**
For octoprint to grow the user base needs a simple way to stay up-to-date. Docker helps a lot, volumes for configuration also help. This is the missing bit for total unattended updates.

rvalle, 2021-01-08:
actually I restarted the process again with the AUTOMIGRATE enabled and it worked. (from 1.4.8 to 1.5).

I guess if there is a configuration from a previous version there and AUTOMIGRATE is disabled, would be better to just stop and display a warning to the user.

LongLiveCHIEF, 2021-01-10:
The problem with stopping is that there's no way to fully test all possible outcomes of the AUTOMIGRATE feature to ensure it won't be non-destructive, and this would force the user to migrate.

The images are versioned using semantic versioning, but the tags for the images follow upstream, and this means you should always check octoprint-docker releases before pulling to see if you're going to be pulling a breaking change and prepare accordingly.

I _could_ add a _IGNORE_ env var of some kind to pair with the behavior you suggest to allow users to override and start without AUTOMIGRATE, but it might be more trouble than it's worth.  The v2 file structure was only around for a few weeks, and the you're only the second user to report an impact since v3 was released, so ATM I would assign a higher priority to other features.

This is still something to consider though for future-proofing due to the container semantic versioning being different than the octoprint product, so I'm going to mark it as an enhancement and add it to the backlog.

ElectricImpossible, 2021-01-10:
I have been burnt by this tonight. I had hoped by using 
`-e AUTOMIGRATE=true`
I would get my settings back, no luck for me and had to use a previous backup :(

LongLiveCHIEF, 2021-01-10:
> I have been burnt by this tonight. I had hoped by using
> `-e AUTOMIGRATE=true`
> I would get my settings back, no luck for me and had to use a previous backup :(

@ElectricImpossible Glad you had a backup. You're exactly the scenario I've been waiting to hear from so I can further enhance the testing for AUTOMIGRATE. Can you tell me more about what settings you lost and how you're mounting volumes? I'm wondering if there could have been any possible actions that would have wiped your config and plugin dirs prior to running the automigration.

ElectricImpossible, 2021-01-10:
I have probably jumped the gun here

Im using unraid to host my container and was only mounting /home/octoprint out. Upon playing around it seems I never mounted /octoprint/octoprint and thus it was left behind in the previous container.

Let me see if I can get my mountings straight and happy to be a Guinea pig

rvalle, 2021-01-11:
@LongLiveCHIEF My migration worked, but so far I noticed that the FullScren plugin was not there, I had to reinstall.

I follow the docker_compose so I mount -v octoprint:/octoprint

I assumed that plugins were not migrated... and I reinstalled the ones I typically use.

closed 2021-02-26 by LongLiveCHIEF