OctoPrint/octoprint-docker #171: Automatic update within Docker

dzarda, 2021-01-18:
**Describe the bug**
An automatic update process within the Octoprint UI failed. Considerin this is a dockerized distribution, is this even supposed to work?

**Container Details**
Octoprint 1.4.2 running on Raspbian.

**To Reproduce**
Steps to reproduce the behavior:
1. Start on a 1.4.2 version docker image
2. Use the automatic update process suggested within the UI
3.  docker-compose restart octoprint
4. Browser page stuck at loading

**Expected behavior**
A successful update was expected.

**Screenshots**
*UI being stuck at loading:*
![obrazek](https://user-images.githubusercontent.com/3955056/104932937-896e0a00-59a8-11eb-8cd9-186ae9369b27
.png)

**Desktop:**
 - OS: [Windows]
 - Browser [Firefox]
 - Version [84.0.2]

**Logs:**
[octolog.txt](https://github.com/OctoPrint/octoprint-docker/files/5831080/octolog.txt)

LongLiveCHIEF, 2021-02-19:
Usually it _should_ work, but we've run into issues a few times with this specific version of octoprint. We recently made backup and restore via the UI possible in the docker image, so I would suggest for now that you run a backup, pull the new image, check out the [CHANGELOG.md](https://github.com/OctoPrint/octoprint-docker/blob/master/CHANGELOG.md) for the last 2 _image_ relase notes about setting `AUTOMIGRATE=true`.

The 1.4 images were mostly v2 image versions, and now we're on v3 of the image which has some changes intended to support upgradeability and backup/restore functionality.

LongLiveCHIEF, 2021-02-26:
@dzarda any updates?

dzarda, 2021-03-06:
Sorry for the delay and thanks for getting back to me.

What I did then was a clean reinstall. I'll study the AUTOMIGRATE functionality more.

closed 2021-03-07 by LongLiveCHIEF