OctoPrint/octoprint-docker #172: Q: How to add plugins when building docker image

NeverBehave, 2021-01-22:
## Question/Problem Description

i am using `octoprint:1.4.2` for a long time, and it looks like recently that image was updated and somehow my plugin disappeared when using the updated base. Also, it seems that plugins like `bedmeshvisualizer` won't installed properly --- showing `install unknown` after completion and missing `numpy` after restart.

### System Details

`Dockerfile` file:

```Dockerfile
FROM octoprint/octoprint:1.4.2

# Install plugins
RUN pip install \
    "https://github.com/birkbjo/OctoPrint-Themeify/archive/master.zip" \
    "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/1.0.0.zip" \
    "https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip" \
    "https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip" \
    "https://github.com/Kragrathea/OctoPrint-PrettyGCode/archive/v1.1.zip" \
    "https://github.com/cesarvandevelde/OctoPrint-M73Progress/archive/v0.2.1.zip"
```

 `docker-compose.yml` file:

```
version: '3.7'

services:

  klipper:
    image: klipper
    volumes:
      - /dev:/dev
      - ./klipper-data:/cfg
      - /tmp/klipper:/tmp
    command: ["/cfg/klipper.cfg", "--insecure-tty", "-l", "/dev/stdout"]
    privileged: true
    restart: unless-stopped

  octoprint:
    depends_on:
      - klipper
    image: octoprint
    ports:
      - 5000:5000
    privileged: true
    volumes:
      - /dev/pts:/dev/pts
      - /tmp/klipper:/tmp/klipper
      - ./octoprint-data:/octoprint
    restart: unless-stopped
```

Include the output of `docker version` (and `docker-compose` version if using compose):

```
docker-compose version 1.27.4, build unknown
docker-py version: 4.3.1
CPython version: 3.8.6
OpenSSL version: OpenSSL 1.1.1h  22 Sep 2020
```

## Describe your hardware

running on raspberry pi 4 64bit Ubuntu

LongLiveCHIEF, 2021-02-02:
Try this:

```
FROM octoprint/octoprint:1.4.2

ENV PIP_USER false

# Install plugins
RUN pip install \
    "https://github.com/birkbjo/OctoPrint-Themeify/archive/master.zip" \
    "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/1.0.0.zip" \
    "https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip" \
    "https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip" \
    "https://github.com/Kragrathea/OctoPrint-PrettyGCode/archive/v1.1.zip" \
    "https://github.com/cesarvandevelde/OctoPrint-M73Progress/archive/v0.2.1.zip"
    ```

LongLiveCHIEF, 2021-02-19:
closed in favor of #22

closed 2021-02-19 by LongLiveCHIEF