OctoPrint/octoprint-docker #183: [BUG] Updating OctoPrint in Docker never boots after upgrading

MohammedNoureldin, 2021-04-01:
**Describe the bug**
OctoPrint in Docker works properly until I upgrade to the next OctoPrint version, it never boots again after restarting. This problem has been there since my first day using OctoPrint in Docker (since November I think). And I am using the latest Docker image. The container is hosted on a Raspbian on RPi3B+

Checking the log of the docker container shows the following:

```
2021-04-01 20:22:16,012 - octoprint.server.util.flask - INFO - Passively logging in user ME from 192.168.1.123
2021-04-01 20:22:16,013 - octoprint.access.users - INFO - Logged in user: ME
2021-04-01 20:22:16,155 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/site-packages/octoprint/static/js/lib/jquery/jquery-ui.js'
2021-04-01 20:22:16,217 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/site-packages/octoprint/static/js/lib/hls.js'
2021-04-01 20:22:16,416 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/site-packages/octoprint/static/js/app/viewmodels/loginui.js'
2021-04-01 20:22:16,566 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/site-packages/octoprint/static/vendor/font-awesome-5.15.1/css/all.min.css'
2021-04-01 20:22:16,570 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/site-packages/octoprint/static/vendor/font-awesome-5.15.1/css/v4-shims.min.css'
2021-04-01 20:22:16,818 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/site-packages/octoprint/static/vendor/font-awesome-5.15.1/css/all.min.css'
2021-04-01 20:22:16,824 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.8/site-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
[... 60 lines cut ...]
      - MJPG_STREAMER_INPUT="-y -n -r 1280x1024"

volumes:
  octoprint:
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
OctoPrint to work properly after update and restart.

**Screenshots**
It hangs on this page:
[image]

**Desktop (please complete the following information):**
This is the system I am using to browse OctoPrint from
 - OS: Windows 10
 - Browser Chrome
 - Version 89.0.4389.90 (Official Build) (64-bit)

swindt, 2021-05-13:
Same issue here.. Running the dockerized octoprint 1.4.2 on an RPI 4 works fine. But updating it to 1.6.1 somehow corrupts the container.

Paraphraser, 2021-05-13:
I think you might misunderstand how containers work.

When you first set up your `docker-compose.yml` and then run:

```
$ docker-compose up -d
```

Docker pulls the `octoprint/octoprint` image from DockerHub. That image is then used to instantiate the running container.

While the container is running, it is free to make changes anywhere within its own file system.

If you stop everything:

```
$ docker-compose down
```

then any changes made inside the container while it was running are lost, **except** for anything in (or below) the internal path:

```
/octoprint
```

That's because that internal path is the right hand side of a `volumes` directive:

```
- octoprint:/octoprint
```

Anything written inside the container at `/octoprint` (right hand side) is preserved outside the container at `octoprint` (left hand side).

When the container next spins up, the **only** changes that are preserved from the previous run at those in `octoprint:/octoprint`.

A collection of one or more `volumes` statements is generally called the "persistent storage area", implying that all other changes are non-persistent.

Now, suppose you do what you said you did. It's reasonable to assume that step 2 ("configure Octoprint") is making changes within `/octoprint` inside the container. Those changes will persist across a container down/up.

It is also reasonable to assume that step 4 makes changes to other things like, say:

```
/usr/local/bin/octoprint
```

but those changes will **not** persist across a container down/up because the next instantiation of the container is always built from the image, and the image isn't affected by step 4.

So, assume, for the moment, that what is going on here is that step 4 is making interdependent companion changes in both `/octoprint` and somewhere else inside the container. From OctoPrint's perspective, it's just installing some new binary or Python file and updating a config file or equivalent to match. It thinks that nothing will change across a restart but it is wrong. When the container spins up again, it's the old "binary" but the new config file, and the old "binary" has no idea what the new config file means.

> Keep in mind this is all speculation. I don't actually **know** that that is what is happening.

The **correct** way to update a Docker container is:

```
$ docker-compose pull
$ docker-compose up -d
$ docker system prune
```

The `pull` looks at:

```
image: octoprint/octoprint
```

in `docker-compose.yml` and interprets it as:

```
image: octoprint/octoprint:latest
```

It goes looking on DockerHub, pulling down any newer image with the "latest" tag.

The `up` then builds a new container from the new image, and performs a new-for-old swap of the running containers at the very last moment. There's barely any downtime.

The `prune` does a number of things but, in relation to a `pull`, it means "get rid of the older image".

This doesn't **guarantee** that something similar to the current problem won't happen. It's entirely possible that you'll pull down version 1.6.1, spin it up, and it will freeze because it expects something in the persistent storage area (`/octoprint` inside the container) to have already been set up during the update.

If that happens, the usual trick is to take the container down, erase the persistent storage area (the left hand side of `octoprint:/octoprint`), bring the container up again, and let it re-initialise the persistent storage area.

Last point. One of the cool things about Docker is how easy it is to go backwards. If you edit `docker-compose.yml` to be:

```
    image: octoprint/octoprint:1.6.0
``` 

and then do an "up", the 1.6.0 image will be pulled down and your OctoPrint service will be pinned to that version. That doesn't stop you from doing a "step 4" and mucking things up but it's really useful if, say, 1.6.1 has a bug which prevents it from working at all.

Generally, between pinning and erasing persistent storage areas, you can usually get containers going again in a jiffy.

MohammedNoureldin, 2021-05-13:
@Paraphraser, so to do update I need to remove my old volumes completely?

Paraphraser, 2021-05-14:
Assuming your current state of play is:

1. Image is 1.6.0
2. You did "step 4" and updated that internally, with companion changes in volumes.

Then just doing a:

```
$ docker-compose down
$ docker-compose pull
$ docker-compose up -d
```

will probably work. The "down" will get rid of the hung v1.6.0 container. The "pull" will download the 1.6.1 image. The "up" will spin up a new container based on the 1.6.1 image, and the new container will probably be happy with the volumes.

But, if a down/pull/up still freezes then, yes, take it down, erase the volumes, and bring it up again.

Absolute worst case is pin to 1.6.0 (or earlier - I'm running 1.6.0 OK), pull, erase the volumes, and up.

Paraphraser, 2021-05-14:
For the record, I just did a "pull" and "up" on my own system. It was running 1.6.0. It is now running 1.6.1. No drama at all. Just worked.

LongLiveCHIEF, 2021-06-25:
well... not sure if that makes me feel better or worse. Going to close for now, we seemed to have some sporadic reports of this prior to 1.6.0, and it wasn't just docker images it was happening to.

I'll be keeping an eye out though for future reports of this to see if I need to prioritize writing an "upgrade" e2e test in the pipeline.

closed 2021-06-25 by LongLiveCHIEF