OctoPrint/octoprint-docker #185: Updating Container delete all files from bind folder

Sonne1989, 2021-04-07:
**Container Details**
2021-02-27T15:26:38.103Z

In Octoprint you have the possibility to Upload the GCode via Webinterface upload button. I decided to go another way. 
I save my sliced GCode files directly to a folder (mounted network drive) on the server. So i bind the folder to container. I see all my files an can work with them. If i update my container, e.g. add a new device, all my files in the binded folder was delete. 

Desktop OS: OSX 10.15.6
Browser: Safari
Server OS: Unraid

LongLiveCHIEF, 2021-04-08:
This doesn't sound like a bug, just sounds like normal docker behavior.  Can you share the way you were mounting folders by sharing the docker commands or the docker-compose configs?

Sonne1989, 2021-04-08:
docker run -d --name='OctoPrint_MK3s' --net='br0' --ip='192.168.3.6' -e TZ="Europe/Berlin" -e HOST_OS="Unraid" -e 'TCP_PORT_5000'='80' -e 'ENABLE_MJPG_STREAMER'='true' -e 'CAMERA_DEV'='/dev/video2' -e 'MJPG_STREAMER_INPUT'='-n -r 1024x768' -v '/mnt/user/appdata/octoprint/mk3s':'/home/octoprint':'rw' -v '/mnt/user/3DDruck/3DDruck/':'/octoprint/octoprint/gcode':'rw' --device='/dev/ttyACM0' --device='/dev/video2:/dev/video2'  -h octoprintmk3s 'octoprint/octoprint:latest

Sonne1989, 2021-04-08:
At this moment i see the Problem. I'm so sorry. I'm mounted the /octoprint/octoprint/gcode folder. But this folder is not persistent. when i take the /home/octoprint folder all is good.

LongLiveCHIEF, 2021-04-08:
The `/home/octoprint` folder is not the right folder either.

closed 2021-04-08 by LongLiveCHIEF

Sonne1989, 2021-04-08:
Which Folder should i use?

LongLiveCHIEF, 2021-04-08:
`/octoprint`