OctoPrint/octoprint-docker #187: [BUG] Several plugins do not work due to noexec on /octoprint mountpoint

inigoml, 2021-04-25:
**Describe the bug**
Some plugins do not work. One of them is Octoprint-telegram or Octoprint-bedlevelvisualizer.

**Container Details**
2021-02-27T15:26:38.103Z

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings->Pluggin Manager.'
2. Add serveral plugins, among them, Telegram and BedLevelVisualizer.
3. Let octoprint restart
4. These plugins do not appear in Plugin Manager (but yes other plugins).

Looking at octoprint.log we obtain this trace:

2021-04-25 18:06:33,386 - octoprint.plugin.core - ERROR - Error loading plugin telegram
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_telegram/__init__.py", line 3, in <module>
    from PIL import Image
  File "/octoprint/plugins/lib/python3.8/site-packages/PIL/Image.py", line 109, in <module>
    from . import _imaging as core
ImportError: /octoprint/plugins/lib/python3.8/site-packages/PIL/_imaging.cpython-38-x86_64-linux-gnu.so: failed to map segment from shared object
2021-04-25 18:06:33,410 - octoprint.plugin.core - ERROR - Error loading plugin bedlevelvisualizer
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/octoprint/plugins/lib/python3.8/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/octoprint/plugins/lib/python3.8/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: /octoprint/plugins/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-x86_64-linux-gnu.so: failed to map segment from shared object

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 9, in <module>
    import numpy as np
  File "/octoprint/plugins/lib/python3.8/site-packages/numpy/__init__.py", line 145, in <module>
    from . import core
  File "/octoprint/plugins/lib/python3.8/site-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
[... 83 lines cut ...]
  * The Python version is: Python3.8 from "/usr/local/bin/python"
  * The NumPy version is: "1.20.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: /octoprint/plugins/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-x86_64-linux-gnu.so: failed to map segment from shared object

**Expected behavior**
Plugin should be loaded and work.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Firefox
 - Version 

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22] 88

**Additional context**
I think this problem is due to numpy package.
octoprint volume is mounted as "noexec", and in order to work should be mounted as "exec".

LongLiveCHIEF, 2021-05-24:
the bed level visualizer is one of the plugins I use for e2e testing this image.  The plugins path should be /octoprint/plugins and not /octoprint.  Can you share your docker-compose or docker run command?

inigoml, 2021-05-26:
I stopped using Octoprint docker on x86 and now I'm using OctoPi. Sorry.
Bed level and Telegram plugins were installed from GUI. Anyway, my docker-compose is still at server:

root@nas-inigoml:~/octoprint# cat docker-compose.yaml
```
version: '2.4'

services:
  octoprint:
    image: octoprint/octoprint
    restart: unless-stopped
    ports:
      - 8081:80
    devices:
    #  - /dev/ttyACM0:/dev/ttyACM0
    #  - /dev/video0:/dev/video0
        - /dev/ttyUSB0
    volumes:
     - octoprint:/octoprint
    # uncomment the lines below to ensure camera streaming is enabled when
    # you add a video device
    #environment:
    #  - ENABLE_MJPG_STREAMER=true

  ####
  # uncomment if you wish to edit the configuration files of octoprint
  # refer to docs on configuration editing for more information
  ####

  config-editor:
    image: linuxserver/code-server
    ports:
      - 7443:8443
    depends_on:
      - octoprint
    restart: unless-stopped
    environment:
      - PUID=0
      - GUID=0
      - TZ=Europe/Madrid
    volumes:
      - octoprint:/octoprint

volumes:
  octoprint:

```