OctoPrint/octoprint-docker #190: [BUG] MJPG_STREAMER build is exiting on certain webcams  "046d:0991 Logitech, Inc. QuickCam Pro for Notebooks"

jzhvymetal, 2021-05-24:
For some reason the current build in the docker of MJPG_STREAMER exits when using  "046d:0991 Logitech, Inc. QuickCam Pro for Notebooks" .  Same docker works correctly for "046d:082d Logitech, Inc. HD Pro Webcam C920"  with no problem.  This was also happening with standalone OctoPi image for the PI.  I was able to copy the build of MJPG_STREAMER  from the docker image "openhorizon/mjpg-streamer-pi3" and replace into the octoprint-docker and it fixed the problem for the "046d:0991 Logitech, Inc. QuickCam Pro for Notebooks".  Also for some reason I also tried compiling MJPG_STREAMER  from the latest git repo but it also did not work.

Below are the commands that I used for installing both dockers.

##Create a file named /etc/udev/rules.d/99-video.rules 
sudo nano /etc/udev/rules.d/99-video.rules
##Add the following line to that file:
KERNEL=="video*",MODE="0666"
reboot

sudo docker run -d -it --privileged -p 8080:8080 --name=mjpg_streamer --restart=always openhorizon/mjpg-streamer-pi3 ./mjpg_streamer -o "output_http.so -w ./www" -i "input_uvc.so -r 960x720 -f 10 -d /dev/video0"

docker run -d --restart=always -v /dev:/dev --privileged -e ENABLE_MJPG_STREAMER=true -e CAMERA_DEV=/dev/video0 -e MJPG_STREAMER_INPUT="input_uvc.so -r 960x720 -f 10 -d /dev/video0"  -p 80:80 --name octoprint_TEST octoprint/octoprint

LongLiveCHIEF, 2021-05-24:
This sounds like you're not setting up the camera on your host correctly.

LongLiveCHIEF, 2021-05-24:
Can you join the discord and ask about your setup there? this is a support issue, not a bug.

closed 2021-05-24 by LongLiveCHIEF

jzhvymetal, 2021-05-25:
How can this be a support issues when the same camera functions with the same setup with docker that also runs MJPG_STREAMER inside of it?

LongLiveCHIEF, 2021-05-26:
There are a dozen reasons why that could be the case. If I could see the implementation of the openhorizon image, I could tell more, but there's no repo associated with that image that I could find.

You also said that this was a problem with upstream mjpg streamer project, which would indicate at best it would be an issue _there_ and not _here_.