OctoPrint/octoprint-docker #191: Followup to comments on pull request adding OctoPrint/octoprint-docker to SensorsIot/IOTstack

Paraphraser, 2021-05-30:
@LongLiveCHIEF - thank you for taking the time to look through and comment on [SensorsIot/IOTstack pull request #328](https://github.com/SensorsIot/IOTstack/pull/328).

I have taken all your suggestions seriously. My objective is to improve the overall quality of the user experience for IOTstack users who choose to install OctoPrint/octoprint-docker as one of their containers.

Unfortunately, I am having a lot of trouble getting things to work as per your comments. I don't *think* it's an IOTstack issue. I think it's more likely to be either some fundamental misunderstanding on my part, or something peculiar about my setup. However, I am going to need your help to sort it out.

To summarise the stage I am at right now:

1. I can't get either `/webcam` or `/webcam?action=stream` to work for external access to the camera feed.
2. I can't get `/dev/ttyAMA0:/dev/ttyAMA0` to work.
3. When I tried to change the admin password from the CLI using your approach, I could not get it to work.

The rest of this issue is a blow-by-blow of my attempts thus far.

<hr>

My setup is:

* WiFi-connected Raspberry Pi 3B+ with the host name "octopi".
* Raspberry Pi camera (ribbon cable into the Pi).
* AnyCubic I3 Mega 3D printer.
* USB cable between the two.

I only have the one printer so I lack the means to test other makes and models.

The starting point for what follows is:

* basics:

	```
	$  cat /proc/device-tree/model
	Raspberry Pi 3 Model B Plus Rev 1.3

	$ uname -a
	Linux octopi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux

	$ docker images
	REPOSITORY               TAG       IMAGE ID       CREATED       SIZE
	octoprint/octoprint      latest    3bb0c99d08fb   2 weeks ago   702MB
	
	$ docker exec octoprint octoprint --basedir /octoprint/octoprint systeminfo | grep "octoprint\.version"
	octoprint.version: 1.6.1
	```

* service definition in `docker-compose.yml`:

	```
	  octoprint:
	    container_name: octoprint
	    image: octoprint/octoprint
	    restart: unless-stopped
	    environment:
	      - TZ=Australia/Sydney
	      - ENABLE_MJPG_STREAMER=true
	      - MJPG_STREAMER_INPUT=-r 1152x648 -f 10
	      - CAMERA_DEV=/dev/video0
	    ports:
	      - "9980:80"
	    devices:
	      - /dev/ttyAMA0:/dev/ttyAMA0
	      - /dev/video0:/dev/video0
	    volumes:
	      - ./volumes/octoprint:/octoprint
	```

	This incorporates two of your recommendations:

	1. The omission of the port mapping:
	
		```
[... 259 lines cut ...]
	Password changed for user admin.
	```

8. Logout from UI. Login:

	* "secondPassword" - fails
	* "firstPassword" - succeeds

	Conclusion: Password change has not come into effect.

9. Restart container and re-test:

	* "firstPassword" - fails
	* "secondPassword" - succeeds

	Conclusion: the command syntax and sequence needed to change the admin password from the CLI is:

	```
	$ docker exec octoprint octoprint --basedir /octoprint/octoprint user password --password «new password» «existing username»
	$ docker-compose restart octoprint
	```

Should the single `user add` command have worked? If yes, any idea why it did not work when I tried it?

The `user password` command followed by a container restart *does* seem to work. Is there any reason why I should not document that as the recommended approach?

LongLiveCHIEF, 2021-05-30:
Hey! thanks for opening to follow up!  I'm going to take a moment to look at everything here so I can respond.  If you can, let's chat on discord!

LongLiveCHIEF, 2021-05-30:
I had already planned to do a video on a bunch of these things this morning, so instead of saving it and waiting for the edited version, i decided to livestream it and include timestamps because several of these pieces are relevant.

This video is _not_ meant to be my primary response to your issues, but I hope that several pieces can help further our discussions.  I'll still be going through and responding to each item as needed.

Here's the YouTube video: https://youtu.be/eKboXA0CBvo

The parts I hope you can spend a few minutes checking out for me start at 1:12:45.  Specific timestamps for demo/discussion about /dev/tty, using cli to reset user passwords/add users are in the video description.... but really I'd like your thoughts on anything after 1:20:00 to the end of the video (about the last 25 mins).  those bits are especially relevant to things like IoT stack and octofarm, since they discuss preconfiguration and decoupling usb devices.

All of those bits are the parts I need the most feedback on from systems like those to help me figure out what may need to change in the image, or what the missing documentation or features are.

I'll be responding more to specific issues you brought up over the next 48 hours or so, as some of them may be more than questions and might need to be logged as bugs or features as you and I work together to see what works differently between our systems or why things work on my test farm that don't work for the systems you test on.

Paraphraser, 2021-05-31:
Sorry for the delay. I'm not ignoring you. Had other commitments today. Will get onto it.

LongLiveCHIEF, 2021-05-31:
Totally understand! That's how OSS works.

Paraphraser, 2021-06-01:
### Camera investigations - continued

Spoiler alert: problem solved!

After watching your video, my first thought was that the explanation might lie in the fact that you were mapping port 80 to port 80 while I was mapping port 9980.

That was a blind alley but it helped me to discover the actual problem.

These two webcam suffixes I was trying in the earlier testing were **wrong**:

* `/webcam` (from [commit comment 51240522](https://github.com/SensorsIot/IOTstack/pull/328#commitcomment-51240522))
* `/webcam?action=stream` (from [commit comment 51240575](https://github.com/SensorsIot/IOTstack/pull/328#commitcomment-51240575))

The correct suffix is:

* `/webcam/?action=stream`

I've been kicking myself because I already "knew" this. My own notes include:

```
Stream URL: /webcam/?action=stream
Snapshot URL: http://localhost:8080/?action=snapshot
Path to FFMPEG: /usr/bin/ffmpeg
```

I just didn't spot the missing "/" used throughout the earlier testing because it's a bit too subtle, or I'm a bit too blind, or one ant short of a picnic, or some combo of all three.

Anyway, with:

```
  octoprint:
    container_name: octoprint
    image: octoprint/octoprint
    restart: unless-stopped
    environment:
      - TZ=Australia/Sydney
      - ENABLE_MJPG_STREAMER=true
      - MJPG_STREAMER_INPUT=-r 1152x648 -f 10
      - CAMERA_DEV=/dev/video0
    ports:
      - "9980:80"
    devices:
      - /dev/ttyAMA0:/dev/ttyACM0
      - /dev/video0:/dev/video0
    volumes:
      - ./volumes/octoprint:/octoprint
```

Test:

```
$ curl -G http://127.0.0.1:9980/webcam/?action=stream
--boundarydonotcross
Warning: Binary output can mess up your terminal. Use "--output -" to tell 
Warning: curl to output it to your terminal anyway, or consider "--output 
Warning: <FILE>" to save to a file.
```

Qapla!

Summary:

* Only port 80 needs to be exposed and mapped (as you said)
* The internal configuration (Settings » Webcam & Timelapse) is:

	- Stream URL: `/webcam/?action=stream`
	- Snapshot URL: `http://localhost:8080/?action=snapshot`
	- Path to FFMPEG: `/usr/bin/ffmpeg`

* URLs for reaching the camera from outside the container are:

	- Stream: `http://raspberrypi.local:9980/webcam/?action=stream`
	- Snapshot: `http://raspberrypi.local:9980/webcam/?action=snapshot`

Paraphraser, 2021-06-01:
### Password investigations - continued

I think it's reasonably clear that the password problems were down to just the wrong commands. For IOTstack, I'm going to say words to the effect of "pick a username" then:

* if you forget the username:

	```
	$ docker exec octoprint octoprint --basedir /octoprint/octoprint user list
	```

* if you forget the password:

	```
	$ docker exec octoprint octoprint --basedir /octoprint/octoprint user password --password «new password» «existing username»
	$ docker-compose restart octoprint
	```

Paraphraser, 2021-06-01:
### Device investigations - continued

Based on what you said in the video and my **casual** inspection of 71-octoprint.docker.rules, it looks to me like our approaches are sort of the inverse of each other.

You are attacking the problem by transmitting signals into the container to cause it to react, and giving it permission to do what needs to be done.

> If it were me, rather than trying to shoehorn everything into the udev rule, I think I'd be defining a script which got added to the container at Dockerfile time, then just invoke that script from the udev rule along with any needed parameters.

I'm attacking it using complementary containers and activating the container that's appropriate to the situation.

Mine is messier than yours (four components outside the container) whereas yours only seems to have one (not counting the cgroup rule).

Mine is also a lot slower to react than yours appears to be. The time to tear down the "printer off" container, then bring up the "printer on" container, and wait for OctoPrint to be open for business is a significant chunk of a minute.

On the plus side, my approach turns the camera off when the printer isn't in use. Have you thought about how to implement that kind of feature?

I'll chew on this for a while longer but, for now, my plan is to stick with my own approach and "await developments".

Your explanation of the cgroup rule was very informative.

<hr>

Based on my experiments and your video, I think it's reasonably clear that `ttyAMA0:ttyAMA0` was never going to work because the right hand side needed to be `ttyACM0`. I think we can call that a typo.

Paraphraser, 2021-06-01:
### Odds and ends

A couple of things struck me as I watched your video. They are more Docker than OctoPrint. You probably know some or all of this but what you said as you went along made me think it might be worth making some notes and then writing it all down. Who knows, something might turn out to be useful.

<hr>

First, the essential difference between `docker` and `docker-compose` is that the latter needs a compose file to work. That implies the former can run from anywhere.

In my case, all the action takes place in `~/IOTstack` so any `docker-compose` command either needs:

```
$ cd ~/IOTstack
$ docker-compose ...
```

or the path to the compose file needs to be passed on the command:

```
$ docker-compose -f ~/IOTstack/docker-compose.yml ...
```

I'm a fast typist but I hate typing and I also find the need to remember these command differences really boring. I've defined a bunch of aliases ([Paraphraser/IOTstackAliases](https://github.com/Paraphraser/IOTstackAliases)) and shell functions which, with appropriate adaptation, might help you too. Each alias is basically the result of me making one-too-many mistakes and thinking "never again". 

For example, if I want to bring up the entire stack:

```
$ UP
```

If I want to bring up just one container:

```
$ UP octoprint
```

You said something about "history" (in the sense of repeating previous shell commands) in a way which seemed to imply you weren't sure whether it was there.

The answer is that it depends on whether the container has been recreated. A "restart", which I would do as:

```
$ RESTART octoprint
```

doesn't recreate the container so anything the container has written to its non-persistent storage layer will still be there, including bash history.

The same happens on a `stop` followed by a `start`. The container hasn't gone anywhere so everything is intact.

Recreating the container either needs a "down" of the entire stack:

```
$ DOWN
```

or a "stop" then a "remove" of the stopped container, which I do as:

```
$ TERMINATE octoprint
```

If I want to be 100% sure there's no carryover, I use:

```
$ RECREATE octoprint
```

which has the effect of a `TERMINATE octoprint` followed by an `UP octoprint`.

Also, if you make a material change to a container's service definition in `docker-compose.yml`, and then "up" the container, it will have the effect of terminating the old container and bringing up the new container according to the new service definition. That implies the loss of non-persistent storage.

Getting back to `docker` vs `docker-compose`, there's no need to use the latter for executing things inside the container. These two are synonymous:
[... 99 lines cut ...]

Generally, that's true. But, if executing external commands is a desirable feature for octoprint-docker, there is a way around it. See [executing commands outside the node red container](https://sensorsiot.github.io/IOTstack/Containers/Node-RED/#executing-commands-outside-the-node-red-container). I can't think of any reason why the approach I describe there for Node-RED would not work for octoprint-docker.

You *could* automate the key-generation on container first run but I don't think there's any way around manual interaction for the key exchange between the container and user (ie "pi"). However, once the keys have been exchanged, ssh inside the container can send any command it likes to user "pi" and that includes rebooting.

Personally, I don't see any real need for the OctoPrint UI to be able to reboot or shutdown the Pi. It's just as easy to do it via ssh. This suggestion is more aimed at the possibility of **other** situations where it could be useful if OctoPrint running inside the container could cause things to happen outside the container.

<hr>

I'm not a maintainer of IOTstack. I'm more what you would call an active contributor. With many opinions. That I'm not afraid to share.

<hr>

But for the drill-down into the video stream, I would probably not have noticed this.

```
$ curl -I 127.0.0.1:9980 2>&1 | grep Clacks
X-Clacks-Overhead: GNU Terry Pratchett
```

I'm a Terry Pratchett fan. It made the whole journey worth the candle. 

And my name isn't even Art.

References: [Clacks](https://xclacksoverhead.org/home/about) and [Arthur Carry](https://discworld.fandom.com/wiki/Arthur_Carry).

closed 2021-06-01 by LongLiveCHIEF

Paraphraser, 2021-06-17:
Following on from your feedback above and your YouTube video stream, I have completely revised [the gist](https://gist.github.com/Paraphraser/650eeab26794dc7363b7123b33f671c4#printerMajorDevNum).

Instead of swapping profiles, any change of external device is sensed by a udev rule (as before), and then a reaction script inside the container is called. I added a volume mapping of `/dev:/host/dev:ro` so that the container had visibility into the host's devices and could then play "follow the leader", with ability to control the device gained via a cgroup rule, as per your example.

In my quest to have the camera follow the printer, I also found it necessary to alter the s6 mjpg-streamer run file to stop it from looping continuously when the camera hadn't actually been mapped by the reaction script.

The gist provides a choice of three udev rule sets:

1. A slightly simplified variant of your example which will match on any ttyUSBn or ttyACMn (I got it down to one line).
2. A version that will match on a printer's identifiers but still mount as ttyUSBn (tested) or ttyACMn (not tested).
3. A version that goes one step further and associates a human-readable device name with the printer.

While I agree that #1 gets the job done, I actually prefer #3.

I made the "camera follows the printer" support optional so that users could choose between never on, always on, or "follow the printer".

I'd welcome feedback and also your considered reaction to these changes. If you think the modifications to  s6 mjpg-streamer run file and/or the printerDidChange reaction script should be added to your repository so they are common to all builds, I'd be happy with that. Conversely, if you don't think that's appropriate then I'll continue on my current path of eventually incorporating this into the IOTstack template.

LongLiveCHIEF, 2021-06-17:
I believe the modifications you suggested to s6 run script for mjpg-streamer are actually on the backlog, so I would welcome those contributions!

The rest of it sounds really awesome, and I'll look more closely this weekend. I also prefer a more human-readable device name, and was tinkering with trying to find a set of udev rules that would match any printer and map it to `/dev/octoprinter` on the host. 

There are so many though, that that task became somewhat onerous.  If we can accomplish it though, I'd still like to find a way to include that in this repo.

Nice work!

LongLiveCHIEF, 2021-06-17:
the mjpg_streamer run modifications related issue: https://github.com/OctoPrint/octoprint-docker/issues/163

LongLiveCHIEF, 2021-06-25:
First, a disclaimer:  I work on this project as the sole maintainer, and I get about 1-3 hours a month to dedicate to it.  

Keep that in mind when giving feedback.  While we all want professional product quality, when dealing with open source, it is somewhat of an expectation that users looking to contribute will be able to do some learning on their own in the case they are not familiar with the tools used.

That being said, _yes_... the CONTRIBUTING is absolutely out of date. It's really not even relevant anymore since we switched to GitHub actions.

I have a lot of documentation noted on the [roadmap](https://github.com/OctoPrint/octoprint-docker/projects/4), and have been doing what I can to effectively prioritize and balance my time between documentation.  

Now for the part you're not going to like...

I've made every attempt I can to work with you in real-time. It's clear that you are new to docker, and have advised your users to do things that are really bad ideas, or downright security issues.  I asked to chat with you and you basically told me "how dare you assume we're in the same time zone".  

You've lectured me at every opportunity you have, instead of trying to chat with me as you're figuring things out (and it's clear you're learning, because as you yourself have mentioned you figured things out after poking around for a while).

Despite all that, I get stuff like this:

>Other people may be happy to play this game but it's not something I'm going to do again. As far as I'm concerned, I have the answer to the question I first thought of:

>>"are my changes better off in octoprint-docker or IOTstack?"

>For me, IOTstack is the clear winner. It's aimed at the RPi so I don't have to worry about other architectures. I can easily test changes on the RPi, for the price of a local Dockerfile and a "build" that completes within a few seconds.

Well... that's great. The whole point of docker is that it is supposed to not be tied to a specific OS/hardware. So I don't have the luxury of just saying "it works on RPi".  

This is an open source product, not a business.  We fully acknowledge things are missing... and that's why we asked for help in the first place.  Instead of helping to make it better, you get angry and unpleasant because our doc doesn't teach you how to use one of the oldest and well documented build systems in the world. 

This is a product we've put our own blood sweat and tears into and offered it for free. I attempted to help you grow and improve  _your own piece of work even_, and you've been condescending and rude to me at pretty much every interaction.

> The main reason I've gone to the trouble of documenting my experience is so that you are aware that there are a lot of kinks in the test process. I hope something in this proves useful.

There is nothing you have shared that is not already documented and on our roadmap. What you've been documenting is your own journey learning how docker works.

I wish you the best.. but also ask that you consider the nature and spirit of open source before you post. I know it's frustrating to work at something for so long, and that makes it easier to vent than it does to be kind and respectful.

We love contributions, we know things are missing, and we want to work with our users and others to fill those gaps... but that means we all need to share an attitude of learning and improvement.