OctoPrint/octoprint-docker #194: [BUG] Cannot pre-install plugins with custom Dockerfile

Popeyef5, 2021-06-22:
**Describe the bug**
Following the best practices to pre-install a plugin, I created a new Dockerfile which builds from octoprint/octoprint and pip installs the plugin (I've tried both copying the source files into the image and pip installing from github). The problem is that although the console logs the plugin as correctly installed, it's nowhere to be found

**Container Details**

please run `docker inspect --format '{{ index .Config.Labels "org.opencontainers.image.created"}}' octoprint/octoprint:<tag_you_are_using>' and list the date returned.

**To Reproduce**
Steps to reproduce the behavior:
1. Create Dockerfile
2. inherit FROM octoprint/octoprint
3. RUN pip install https://github.com/FractalManufacturing/OctoPrint-FractalBridge/archive/master.zip
4. with Dockerfile ready, create simple compose file, sudo docker-compose build and then up
5. plugin is nowhere to be found

**Expected behavior**
The plugin should be preinstalled.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

LongLiveCHIEF, 2021-06-25:
@Popeyef5 in your `Dockerfile`, can you try adding `ENV PIP_USER false` before running your `pip install`?

Also, can you share what best practices you're referring to just so I can be sure we're thinking of the same document?  (I haven't published docker specific best practices yet afaik)

LongLiveCHIEF, 2021-08-29:
If @Popeyef5 responds, will reopen issue.

closed 2021-08-29 by LongLiveCHIEF

shaunmulligan, 2021-09-22:
I had the same issue as @Popeyef5 and disabling the `PIP_USER` just for the plugin install seems to have worked. I did the following in my dockerfile:
```
ENV PIP_USER false
RUN pip install -r initial-plugins.txt
ENV PIP_USER true
```
Where `initial-plugins.txt` is just a text file with all my plugins listed.