OctoPrint/octoprint-docker #196: Q: How do I start / stopwebcam?

redjab, 2021-07-01:
## Question/Problem Description

I'm trying to start /stop the webcam stream. I have something like this right now in config.yaml:

```
system:
  actions:
  - action: streamon
    command: s6-svc -r /var/run/s6/services/mjpg-streamer & /etc/services.d/mjpg-streamer/run
    confirm: false
    name: Start video stream
  - action: streamoff
    command: s6-svc -d /var/run/s6/services/mjpg-streamer
    confirm: false
    name: Stop video stream
```

1. The streamoff command works just fine, but the streamon command just stalls the whole UI until I restart the container. 
2. If I `docker exec -it bash` into the container and execute the streamon command manually, then it works. However, afterward when I run the streamoff command again it doesn't stop the stream anymore.

### System Details

I'm running octoprint docker using the following `docker run` command or `docker-compose.yml` file:

```
version: '3.8'
services:
  octoprint:
    container_name: octoprint
    image: octoprint/octoprint
    restart: unless-stopped
    ports:
      - 81:80
    devices:
     - /dev/ttyUSB2:/dev/ttyUSB2
     - /dev/video0:/dev/video0
    volumes:
      - /persist/octoprint:/octoprint
    environment:
     - ENABLE_MJPG_STREAMER=true
    ```

Include the output of `docker version` (and `docker-compose` version if using compose):

```
docker-compose version 1.25.5, build 8a1c60f6
docker-py version: 4.1.0
CPython version: 3.7.5
OpenSSL version: OpenSSL 1.1.0l  10 Sep 2019```

redjab, 2021-07-01:
Nvm I'm dumb. Use -u instead of -r folks

closed 2021-07-01 by redjab

aguinaldoabbj, 2022-08-09:
@redjab Sorry to dig this up. How are you supposed to run those streamon and streamoff commands?

redjab, 2022-08-11:
@aguinaldoabbj  You put this in your octoprint `config.yaml` file
```
system:
  actions:
  - action: streamon
    command: s6-svc -u /var/run/s6/services/mjpg-streamer
    confirm: false
    name: Start video stream
  - action: streamoff
    command: s6-svc -d /var/run/s6/services/mjpg-streamer
    confirm: false
    name: Stop video stream
```

Restart octoprint, then open up the web interface. Next to the settings icon there should be a "power" icon, and those 2 commands should be under there

aguinaldoabbj, 2022-08-11:
Right! Thanks. What about the mjpg-streamer parameters? It will read the env variables defined in dockerfile/docker-compose ?

redjab, 2022-08-12:
@aguinaldoabbj You can find more information here https://github.com/OctoPrint/octoprint-docker/blob/master/README.md. It enables webcam stream within the docker container.