OctoPrint/octoprint-docker #197: [Docker] Older version scheme breaks auto-updaters

Ornias1993, 2021-07-01:
Currently the dockerhub containers an old version scheme following the following format:
5512.0.0

Basically:
Highnumer.0.0

This leads to a situation where auto-updating systems like Renovate, will assume those old version numbers are newer than the actual latest release. (as 5512.0.0 is a higher version number than 1.6.1).

I would request those versions to be removed or retagged in such a way that they won't cause issues with updating systems.

Note:
Moved from https://github.com/OctoPrint/OctoPrint/issues/4183

LongLiveCHIEF, 2021-08-05:
That got published by mistake somehow, it was never a scheme. We follow semver.  I will go through and clean all those out though... thanks for bringing this to our attention!

LongLiveCHIEF, 2021-08-05:
Cleaned up tags.

closed 2021-08-05 by LongLiveCHIEF