OctoPrint/octoprint-docker #203: Q: How do I update my octoprint docker config without deleting my octoprint settings?

amarotica, 2021-09-02:
**The background:** 
I'm on Raspberry Pi 4 and I had an error in my docker-compose.yaml file where the number of spaces before the _device:_ argument was one too many. Apart from not being able to connect to the printer due to the serial port not showing up, octoprint on docker was running great and I had setup all my user accounts and other settings.

**The f*** up:**
I fixed the _device:_ line spacing error in docker-compose.yaml and ran _docker-compose up_ which promptly rebuilt my octoprint instances and deleted all the user accounts and settings I had spent a half hour configuring. It fixed the error I had and the serial port now shows up as intended, but I had to rebuild all my settings.

**The question:**
How do I rebuild (update the octoprint image with the fixed docker-compose.yaml file) and NOT delete the entire contents of it? Shouldn't enabling serial ports be simpler than that?

Thanks for any advice you might have - I'm planning on backing up my SD card now so this doesn't happen again but I'd like to know where I messed up the commands.

Andoramb, 2021-09-28:
can you share your docker-compose.yaml file?

LongLiveCHIEF, 2021-11-25:
There is a whole section on this in the README, and there's even a companion service in the docker-compose.yaml file for this exact purpose.

closed 2021-11-25 by LongLiveCHIEF

ghostsquad, 2022-05-07:
the mounted volume is what this is supposed to handle, correct? so you can destroy the containers but the volume persists?

LongLiveCHIEF, 2022-05-09:
@ghostsquad Yep, exactly.