OctoPrint/octoprint-docker #205: [BUG] Backup & Restore stores incorrectly when called from the command line

1337joe, 2021-09-23:
**Describe the bug**
The built-in "Backup & Restore" plugin stores to the wrong location when called directly through the plugin subsystem instead of using it through the web ui. Additionally, the zip create by calling backup from the command line does not contain most of the backup data that it should.

**Container Details**
[tty2 ~]$ docker inspect --format '{{ index .Config.Labels "org.opencontainers.image.created"}}' octoprint/octoprint:latest
2021-06-06T14:17:51.966Z

**To Reproduce**
Steps to reproduce the behavior:
1. Launch a new octoprint container:
`[tty2 ~]$ docker run --rm -t --name octoprint -p 5000:5000 octoprint/octoprint:latest`
2. Use the web interface to create a user and set up the server. This doesn't change the results, but it makes it easier to inspect and compare to the expected case: http://localhost:5000
3. In a different terminal call the backup plugin to create a backup:
`[tty3 ~]$ docker exec $(docker ps -qf "name=octoprint") octoprint plugins backup:backup`
 - See expected behavior 1 for explanation of first problem.
4. In the web interface navigate to Settings > Backup & Restore. Note that the backup created above is not listed (expected behavior 2)
5. Create a new backup by clicking on "Create Backup Now", note that a backup of ~60KB is now visible in the top portion of the panel.
6. Copy the original backup file into the directory that the web interface puts it in:
`[tty3 ~]$ docker exec $(docker ps -qf "name=octoprint") cp /root/.octoprint/data/backup/octoprint-backup-20210923-173524.zip /octoprint/octoprint/data/backup/`
7. Refresh the web interface and navigate back to Settings > Backup & Restore.
8. Note that there are now two zip files listed (the one created from the command line and the one created through the web ui, but the sizes are different (Expected behavior 3)

**Expected behavior**

*Expected behavior 1:* The backup is written to a path within the docker volume (`/octoprint').
Actual behavior (log output):
`[tty3 ~]$ docker exec $(docker ps -qf "name=octoprint") octoprint plugins backup:backup
Initializing settings & plugin subsystem...
Creating backup at octoprint-backup-20210923-173524.zip, please wait...
Done.
Backup located at /root/.octoprint/data/backup/octoprint-backup-20210923-173524.zip'

*Expected behavior 2:* Closely related to behavior 1, the backup created by calling the plugin directly does not appear in the Settings > Backup & Restore section of the web ui where I would expect to see it.

*Expected behavior 3:* I would expect the backup plugin to create full backups when called through both paths, but what actually happens is that the backups created by calling the backup plugin directly are nearly empty: in my test run the web ui-generated backup is ~60KB while the command line one is only 387 bytes.

**Desktop (please complete the following information):**
The server behaves the same way on both machines I've tried it on without needing any web interaction to see the incorrect path, but my test environments were:
CentOS 8 / Podman 3.2.3 / Firefox 92.0
Arch Linux / Docker 20.10.8 / Firefox 92.0

**Additional context**
As an additional data point, the `nunofgs/octoprint:alpine` image (which is an older version of octoprint) behaves as expected (backup is 61KB and shows in the "Backup & Restore" section of the web interface with the following console output:

`[tty3 ~]$ docker exec $(docker ps -qf "name=octoprint") octoprint plugins backup:backup
Initializing settings & plugin subsystem...
Creating backup at octoprint-backup-20210923-172011.zip, please wait...
Done.`

1337joe, 2021-09-23:
My goal was to set up a script on my server that would back up all my containers to a remote drive. To offer a workaround to anyone else who runs into this:
The [backup scheduler](https://plugins.octoprint.org/plugins/backupscheduler/) plugin can run the backup (behaves just like manually running it through the web ui), then a script can copy the resulting file out of the container volume.

LongLiveCHIEF, 2021-11-25:
Yeah, any of the command line commands need a `--basedir /octoprint/octoprint` argument added to them, and _then_ they'll work just fine.

closed 2021-11-25 by LongLiveCHIEF